helium/router #250: What is the recommended ratio between owned devAddr and active devices ?

disk91, 2021-03-19:
Hello, 
I want to know what are the best practice in terms of DevAddr reusing ?
How many devAddr own regarding the number of active devices.

evanmcc, 2021-03-19:
transferred it, you don't need to reopen.

macpie, 2021-03-19:
Router is trying to be smart and assign devaddrs based on device's geo-location (see [code here](https://github.com/helium/router/blob/master/src/device/router_device_devaddr.erl#L138)). So if devices are well dispersed you could be able to use a way smaller range than you have devices.

Nevertheless a smaller range could have couple of consequences:

1. For packets,  if more devices share a devaddr ( [code here](https://github.com/helium/router/blob/master/src/device/router_device_routing.erl#L865) ) it could "take longer" as we will need to potentially brute force for more than one device.
2. For offers, a device could be mistaken for another one and in worst cases if that other device is deactivated or org ran out of DC, offer could be dropped.

disk91, 2021-03-21:
Thank you for this response. 
I'm not clear about what Offers are ? can you precise please ?

disk91, 2021-03-21:
Second question, how to monitor response time to take care of "take longer" ?

macpie, 2021-03-23:
1. When a hotspot receive a packet it will offer it to Router. Router then needs to tell ([code](https://github.com/helium/router/blob/master/src/device/router_device_routing.erl#L104)) the hotspot if it wants to buy the packet or not ([Offer definition here](https://github.com/helium/proto/blob/master/src/blockchain_state_channel_v1.proto#L56)).
2. The "take longer" should not really be an issue unless you have tens of thousands of devices sharing the same `devaddr` ([routing functions](https://github.com/helium/router/blob/master/src/device/router_device_routing.erl#L858) are being closely monitored and take on average 1.4 milliseconds)

disk91, 2021-03-24:
thank you it makes sense

closed 2021-03-24 by macpie