helium/router #251: What recommendation for regular device xor filter update ?

disk91, 2021-03-21:
Is there something in the router to manage the xor filter update ? do we have to cron the operation on regular basis ?

michaeldjeffrey, 2021-03-23:
XOR filters check themselves every 10 minutes.
https://github.com/helium/router/blob/master/src/router_xor_filter_worker.erl#L455

You can manually trigger XOR update through cli command.
https://github.com/helium/router/blob/master/src/cli/router_cli_device_worker.erl#L41

`docker exec helium_router router device xor`

disk91, 2021-03-23:
I need one more precision, to get the 10 minutes check, you need to have 
```
ROUTER_XOR_FILTER_WORKER=true
```
right ?

michaeldjeffrey, 2021-03-23:
That's correct. Anything other than `true` will be treated as `false`