helium/router #254: Sate channel not recreated after expiration

disk91, 2021-03-24:
```
[2021-03-24 00:41:04.519] <0.1652.0> [info] [router_sc_worker:create_and_send_sc_open_txn:300] Opening state channel for router: "11awcuSbVURPkXX3FbKC7KF6bgEPRZqqPzv1FTEYABMLttUr13E", oui: 6, nonce: 180, id: <<164,127,239,250,26,175,168,94,126,73,54,53,22,234,88,58,85,161,241,135,217,80,104,219,167,51,88,68,218,168,71,59>>
[2021-03-24 00:41:09.063] <0.1927.4> [warning] [blockchain:can_add_block:768] block doesn't fit with our chain,
                                                block_height: 772685, head_block_height: 772682
[2021-03-24 00:41:34.553] <0.1652.0> [info] [router_sc_worker:create_and_send_sc_open_txn:300] Opening state channel for router: "11awcuSbVURPkXX3FbKC7KF6bgEPRZqqPzv1FTEYABMLttUr13E", oui: 6, nonce: 180, id: <<76,52,77,113,211,140,34,233,179,221,115,120,210,146,4,82,122,35,94,13,91,69,14,83,139,107,117,164,65,35,166,17>>
```
I've changed the following parameters:
```
ROUTER_SC_OPEN_DC_AMOUNT=100000
ROUTER_SC_EXPIRATION_INTERVAL=10000
``` 
it seems that to:
```
ROUTER_SC_OPEN_DC_AMOUNT=50000
ROUTER_SC_EXPIRATION_INTERVAL=5000
```
make it working correctly.
Is that normal ?
What are the max values for these two parameters ?

macpie, 2021-03-24:
- Block not fitting is fine, sometimes Hotspots will gossip few blocks and your Router might not be caught up just yet.
- The state channel open will over commit the amount [see here](https://github.com/helium/blockchain-core/blob/master/src/transactions/v1/blockchain_txn_state_channel_open_v1.erl#L238) set, right now the over commit is set to 2. So if you set your `ROUTER_SC_OPEN_DC_AMOUNT=50000` it will open one for `100000`.
- They are max and min values to  `ROUTER_SC_EXPIRATION_INTERVAL` [see here](https://github.com/helium/blockchain-core/blob/master/src/transactions/v1/blockchain_txn_state_channel_open_v1.erl#L212). Min is currently `15` and max is `10080` (as rewards are tied to state channel close we set a limit of about 1 week).

disk91, 2021-03-24:
Thank you for these precision, 
when using ROUTER_SC_EXPIRATION_INTERVAL = 10000, as at init 2 SC are open and the second one is 2xROUTER_SC_EXPIRATION_INTERVAL; I assume the maximum possible value is 5040 here.

Am I wrong ? If I'm rihgt, it would be better to have the second SC using 1/2 of 2xROUTER_SC_EXPIRATION_INTERVAL, then each of the SC renewed will be up to 10080 as expected.

Correct me if I'm wrong.

macpie, 2021-03-24:
You are right, as Router open 2 state channels the max is indeed `5040`. 

FYI we are running state channels of 120 blocks in all of our environments (dev, staging and prod).

macpie, 2021-03-25:
@vihu thoughts on this ^

closed 2021-04-05 by macpie