helium/router #255: JoinRequests Require Better Deduping

lthiery, 2021-03-25:
There is a condition where a JoinRequest is offered and purchased from two different hotspots. 

If the first hotspot manages to get the JoinAccept to the device, the device will know about the first session (devaddr 15040048 in the screenshot). Meanwhile, Router appears to overwrite the first session with the second session (devaddr 1804004), derived from the second purchase.

The device will be very confused and depending on the logic, may take quite a while to realize it doesn't have a session.

[image]

macpie, 2021-03-25:
This should get fixed with https://github.com/helium/router/pull/253

macpie, 2021-03-29:
Fixed

closed 2021-03-29 by macpie