helium/router #256: Possible FCntUp Bug

lthiery, 2021-03-25:
Router appears to not tolerate a confirmed frame being resent. Router appears to assume that its confirmation is well received by the end-device. It is quite possible for the device to not receive the ACK, in which case the specification says:

"The FCnt is not incremented in case of multiple transmissions of an unconfirmed frame (see NbTrans parameter), **or in the case of a confirmed frame that is not acknowledged**."

[image]

In this example, the `FCntUp = 28` frame is received and acknowledged by Router, but the Device did not receive it. The Device tries to send the frame up again, but it appears that Router makes the frame as "late" instead of a retry on behalf of the device.

macpie, 2021-03-26:
Should be fixed in https://github.com/helium/router/pull/253

macpie, 2021-03-29:
Fixed

closed 2021-03-29 by macpie