helium/router #263: Dedupe on Data Fails

lthiery, 2021-03-31:
A late offer seems to get purchased

[image]

lthiery, 2021-03-31:
I wonder if this was introduced when fixing the fact that the device might not increment its FCnt when sending a confirmed uplink and it fails to receive the downlink confirmation.

In this instance, the reported time between the hotspots is way too tight for an allowable retry (retry 3 seconds after the original uplink is fastest allowable). Secondly, this was _not_ a confirmed uplink, therefore I would consider incrementing the FCnt automatically and straight up refusing anything after. ie: only in the case of a confirmed uplink should you tolerate the replay of a packet and only after at least 3 seconds have elapsed.

michaeldjeffrey, 2021-04-02:
If you can get this to happen more let me know. It would be nice to trace the device while this is happening. 
Remember the time listed is when the hotspot picked up the packet, not when router handled the packet. 
There's a couple late packet scenarios this could be falling into, but without seeing it happen, I can't be sure what exactly is going on.

closed 2021-04-13 by macpie