helium/router #268: Integration returns "unknown" on double lat/long hostpot position fields

disk91, 2021-04-01:
The integration payload returns "unknown" as a string is a field lat/long that should be  double.
Using 0 or out-of-bound double value would be a better choice than changing type for typed languages 

2021-04-01 22:42:59.015  WARN 19128 --- [0.1-8090-exec-6] .w.s.m.s.DefaultHandlerExceptionResolver : Resolved [org.springframework.http.converter.HttpMessageNotReadableException: JSON parse error: Cannot deserialize value of type `double` from String "unknown": not a valid Double value; nested exception is com.fasterxml.jackson.databind.exc.InvalidFormatException: Cannot deserialize value of type `double` from String "unknown": not a valid Double value
 at [Source: (PushbackInputStream); line: 1, column: 367] (through reference chain: fr.ingeniousthings.helium.HeliumCommonMessage["hotspots"]->java.util.ArrayList[0]->fr.ingeniousthings.helium.HntHotspot["lat"])]

disk91, 2021-04-01:
At least, it should be documented

macpie, 2021-04-05:
We will, in the future, add the ability to version the payload so that feature/updates like this can be made without breaking people's integrations.

Fixing docs here https://github.com/helium/router/pull/274

closed 2021-04-05 by macpie