helium/router #289: lpp_decoder_error

RaghavRao, 2021-04-18:
I am sending the reference cayenne lpp packet: `03670110056700FF` as a hex string in python but I see lpp_decoder_error.

Should I not be sending as a hex string?

Console output:
```
  "body": "{\"app_eui\":\"snip\",\"decoded\":{\"error\":\"lpp_decoder_failure\",\"status\":\"error\"},\"dev_eui\":\"snip\",\"devaddr\":\"snip\",\"fcnt\":25,\"hotspots\":[{\"channel\":8,\"frequency\":903.9000244140625,\"id\":\"snip\",\"lat\":snip,\"long\":snip,\"name\":\"snip\",\"reported_at\":1618764283239,\"rssi\":-84.0,\"snr\":13.800000190734863,\"spreading\":\"SF7BW125\",\"status\":\"success\"}],\"id\":\"snip\",\"metadata\":{\"labels\":[{\"id\":\"snip\",\"name\":\"snip\",\"organization_id\":\"snip\"}],\"multi_buy\":1,\"organization_id\":\"snip\"},\"name\":\"snip\",\"payload\":\"MDM2NzAxMTAwNTY3MDBGRg==\",\"payload_size\":16,\"port\":1,\"reported_at\":1618764283239,\"uuid\":\"snip\"}"
```

macpie, 2021-04-19:
Data should not be sent in hex

RaghavRao, 2021-04-20:
Hi I got it working in bytes, but the issue persists. I found that this is because helium does not support the Things Stack V3 data types in it's LPP Decoder:

 * Type                 IPSO    LPP     Hex     Data Size   Data Resolution per bit
 *  Digital Input       3200    0       0       1           1
 *  Digital Output      3201    1       1       1           1
 *  Analog Input        3202    2       2       2           0.01 Signed
 *  Analog Output       3203    3       3       2           0.01 Signed
 *  Illuminance Sensor  3301    101     65      2           1 Lux Unsigned MSB
 *  Presence Sensor     3302    102     66      1           1
 *  Temperature Sensor  3303    103     67      2           0.1 °C Signed MSB
 *  Humidity Sensor     3304    104     68      1           0.5 % Unsigned
 *  Accelerometer       3313    113     71      6           0.001 G Signed MSB per axis
 *  Barometer           3315    115     73      2           0.1 hPa Unsigned MSB
 *  Time                3333    133     85      4           Unix time MSB
 *  Gyrometer           3334    134     86      6           0.01 °/s Signed MSB per axis
 *  GPS Location        3336    136     88      9           Latitude  : 0.0001 ° Signed MSB
 *                                                          Longitude : 0.0001 ° Signed MSB
 *                                                          Altitude  : 0.01 meter Signed MSB
 *
 * Additional types
 *  Generic Sensor      3300    100     64      4           Unsigned integer MSB
 *  Voltage             3316    116     74      2           0.01 V Unsigned MSB
 *  Current             3317    117     75      2           0.001 A Unsigned MSB
 *  Frequency           3318    118     76      4           1 Hz Unsigned MSB
 *  Percentage          3320    120     78      1           1% Unsigned
 *  Altitude            3321    121     79      2           1m Signed MSB
 *  Concentration       3325    125     7D      2           1 PPM unsigned : 1pmm = 1 * 10 ^-6 = 0.000 001
 *  Power               3328    128     80      2           1 W Unsigned MSB
 *  Distance            3330    130     82      4           0.001m Unsigned MSB
 *  Energy              3331    131     83      4           0.001kWh Unsigned MSB
 *  Colour              3335    135     87      3           R: 255 G: 255 B: 255
 *  Direction           3332    132     84      2           1º Unsigned MSB
 *  Switch              3342    142     8E      1           0/1

 * 
 */

macpie, 2021-04-20:
Will take a look

closed 2021-05-26 by macpie