helium/router #293: Clear pending downlinks request as part of the backend downlink response (integration)

disk91, 2021-04-26:
It would be great in the downlink response to get the ability to request a downlink queue cleanup for a given device.
This makes sense when the previously queued message does not make sense or the one you want to push now have a high priority and need to be executed immediately.
I'm about making a clean request from the backend by adding a field in the downlink payload structure to request the console to clear the queue remotely.

macpie, 2021-04-29:
There are multiples ways to implement this:

1. Clear queue via a unique URL provided (kind of like `downlink_url`).
2. Clear queue via integration response (http, mqtt, aws ...)
3. Or both

disk91, 2021-04-29:
In my point of view, the easiest way in term of integration is via the integration response. We just have to add a field like clean: true in it to make this request. This is an advantage when we process message / downlink without knowing the device (like in the field tester project I've created).

Have it accessible through and API is also important when the backend application manage automation and want to execute a such action asynchronously with device communication.

Basically both is better

macpie, 2021-04-30:
Sounds good @disk91. This has been scheduled.

michaeldjeffrey, 2021-06-24:
Feature added in https://github.com/helium/router/releases/tag/2021.06.24.0

closed 2021-06-24 by michaeldjeffrey

disk91, 2021-07-01:
I don't see where it has been included in the related link and I don't know how to use it, the documentation has not been updated.

michaeldjeffrey, 2021-07-01:
Sorry about that @disk91,
In the Router repo.
https://github.com/helium/router#clearing-a-device-queue-from-an-integration

In the Docs.
https://docs.helium.com/use-the-network/console/integrations/json-schema/#downlink-transmitting-data-to-a-device

If there are other places you checked expecting to see it could you let me know?

disk91, 2021-07-01:
Cool, Thank you
I was looking on the page : https://docs.helium.com/use-the-network/console/integrations/http/

Unfortunately, I was expecting a different behavior:
I want to return a Downlink value AND clear the previously pending values.

Could we imagine something like 
downlink_string+__clear_downlink_queue__

To manage a such behavior ?