helium/router #307: Decoded payload not available for dynamic URL Params

michaeldjeffrey, 2021-05-12:
We cannot interpolate values from the decoded payload in URL params. 

If a decoded payload is a JSON object, we should be able to pass `{{ decoded.payload.value }}` as a URL param and get the value from the payload.

arkieguy, 2021-05-12:
Here is the decoder function I'm using:

```
function Decoder(bytes, port) {
  var sensor = {};
  var iStr = String.fromCharCode.apply(null, bytes);
  
   // Get values
  sensor.temp = parseInt( iStr.substr( 1, 6 ) ) / 100;
  sensor.humidity = parseInt( iStr.substr( 7, 5 ) ) / 100; 
  sensor.pressure = parseInt( iStr.substr( 12, 5 ) ) / 100; 
  
  return( sensor );
}
```

It produces this in debug mode:

```
{
  "id": "52a60c6c-6a5a-42ef-bbc8-32fd2494bda0",
  "router_uuid": "35ca3012-13ab-4213-9724-51ce2525bceb",
  "category": "uplink",
  "sub_category": "uplink_integration_req",
  "description": "Request sent to <<\"Debug\">>",
  "reported_at": "1620848415134"
}
Device Information
Integration
Debug Request
{
  "id": "de91e61f-d5d0-4fa9-8f31-c9c21823ed56",
  "name": "Debug",
  "status": "success",
  "decoded_payload": {
    "humidity": 29.78,
    "pressure": 51.48,
    "temp": 73.89
  }
}
```
I would expect to be able to use something like this in the value:

{{decoded.payload.temp}}

jdgemm, 2021-05-19:
Thanks for raising we've added to our backlog.

michaeldjeffrey, 2021-08-02:
I forgot to link this issue in the PR. This was fixed with https://github.com/helium/router/pull/349

closed 2021-08-02 by michaeldjeffrey