helium/router #313: Integration notification in case of device retry

disk91, 2021-05-19:
When a device retry a message that has been received by the router (so when the device did not get the ack). The console send the integration call on every retry. 
This is creating a certain complexity in backend if you want to manage it. The addition of a flag with the retry status would help to filter easily in backend. Enventually the retries could not be pushed to backend as the payload has already been reported.

macpie, 2021-08-18:
The retry payload include the same frame count right?

disk91, 2021-08-24:
Yes it is, but if you want to use that to determine if it is a retry, you need to have a Db request, witch means you need to have a DB, something not always needed on the app standpoint. It is also costly. 
It should be better to get it from the router as this information should be existing.

macpie, 2021-08-24:
> Yes it is, but if you want to use that to determine if it is a retry, you need to have a Db request, witch means you need to have a DB, something not always needed on the app standpoint. It is also costly.
> It should be better to get it from the router as this information should be existing.

Fair enough, this will be implemented has a new feature then

closed 2021-11-29 by macpie