helium/router #316: Apply a DC ratio when user burn HNT to DC for its own account

disk91, 2021-05-24:
As the router activity have larger cost than data transfer, you should consider a router fee for the user.
The idea is to have a ratio like 1.2 = for 12 DC payed, 10 are added to the user account, 2 other are not affected to be consumed for state_channel maintenance and device onboarding (xor). 
This should be a parameter.

macpie, 2021-05-26:
@jdgemm What do you think of this?

jdgemm, 2021-08-12:
sorry for late reply, we've added this to backlog, but will revisit in near future

macpie, 2021-08-18:
@disk91 We are thinking of doing something similar but moving the ratio when the DC are purchased in Console.

Example (number are made up):

Right now `$1 = 10 000DC` with ratio of `1.5` applied `$1 = 5 000DC`.

I believe it will be easier for user to understand then because the ratio of packet DC will stay the same.

mfalkvidd, 2021-08-18:
I think that would be an elegant solution. Except maybe that not only the numbers are made up, the math is also made up. 10,000/1.5 != 5,000 😂

macpie, 2021-08-18:
> fwiiw, I think that would be an elegant solution. Except maybe that not only the numbers are made up, the math is also made up. 10,000/1.5 != 5,000 😂

Fixed

disk91, 2021-08-20:
Sounds good to me @macpie !
Make sure the HNT transfer feature is disabled when ratio != 1

macpie, 2021-08-23:
Seems like we have a good consensus, we will start to spec this out and schedule it. Thank you all for your input

macpie, 2021-09-01:
@jdgemm We should move this issue to Console