helium/router #327: JoinAccept DLSettings should not be always 0

ogimenez-smtc, 2021-05-31:
Dear,

[This line](https://github.com/helium/router/blob/master/src/device/router_device_worker.erl#L1210) always sends `DLSettings = 0` which is an issue at least in US915 and AU915 but maybe in other regions, I did not check (AU923 with dwell time?):

The RX2 datarate is given by the network in the joinAccept frame thanks to the parameter `RX2DataRate`, which is contained in DLSettings:
[image]

As per LoRaWAN specification (1.0.2 to 1.0.4) US915 downlinks are on DR8 to DR13, as Halium network server is providing a value of `RX2DataRate = 0` the device will never receive any downlinks on RX2 datarate.

Note: Some stacks allows the network server to use `RX2DataRate = 0` but downlinks will never work, some others will refuse the joinAccept

pdvaneijk, 2021-05-31:
In the Helium Documentation (https://docs.helium.com/lorawan-on-helium/frequency-plans/#us915) Support for RX2 window downlink is mentioned as supported with the DEFAULT setting of DR8 (SF12 @500Khz) on channel 923.3Mhz but apparently not enabled in the code as Olivier  mentioned above.

pdvaneijk, 2021-06-02:
All, this is still not working for 915Mhz. the Helium Join Accept is not being accepted by the the LoRaWAN stack on my LR1110 based tracker. Unfortunately the Join Accept message payload is not displayed in the Helium console so I cannot decipher if RX2DataRate is non zero or not.

JayKickliter, 2021-06-02:
This was closed automatically. Merged PRs are not immedately shipped to production so it'll take some time to see this running

pdvaneijk, 2021-06-02:
Thanks for explaining. I do have a tracker connected to the staging-console right now. Let me know when this fix is added to this version.

Patrick van Eijk
Semtech Corp
LoRa IoT solutions
(650) 457-4514

***@***.***D59E1F.E929E770]

LoRa Technology - Making our World Smart<https://www.semtech.com/technology/lora>

Warning - External Email

To view our privacy policy, including the types of personal information we collect, process and share, and the rights and options you have in this respect, see www.semtech.com/legal.

ogimenez-smtc, 2021-06-03:
Hi, the fix seems to be in staging now, and I correctly got `MacRx2DataRate= 8` for US915 region, thank you

commit 408c64df (2021-06-02): Expose Region default rx2 datarate index

Fixes #327

Ref: DLSettings from LLS 1.0.4.pdf
https://lora-alliance.org/resource_hub/lorawan-104-specification-package/
PR #331 (2021-06-02): Michaeljeffrey/ch9996/joinaccept dlsettings with as923 updates
Fixes #327

Ref: DLSettings from LLS 1.0.4.pdf
https://lora-alliance.org/resource_hub/lorawan-104-specification-package/
closed 2021-06-03 by ogimenez-smtc