helium/router #337: No AS923 regional plan support for Australia

pdvaneijk, 2021-06-09:
I see that you do not support the AS923 channel plan for Australia ?
 
https://docs.helium.com/lorawan-on-helium/frequency-plans/#us915
 
The AU915 is the old plan folks initially used in Australia, but now all new development is geared towards AS923. You should really only support AS923 on your GTW deployments in Australia. Any device that wants to join will have to run the new As923 stack.

lthiery, 2021-06-09:
@pdvaneijk which AS923 plan in particular is Australia focused on for new development? We will need to update it [here](https://github.com/helium/miner/blob/master/priv/countries_reg_domains.csv#L13) first and foremost as that defines POC behavior which than determines frequencies covered. Our definition of the AS923-x are all [here](https://github.com/helium/miner/blob/30efcb0806fc42a390802357486bfe77ce76fea6/config/sys.config#L108).

pdvaneijk, 2021-06-10:
Our Australian customer says they use band 1: 

For Australia we have been using  AS923_1' => [923.2, 923.4, 923.6, 923.8, 924.0, 924.2, 924.4, 924.6]

closed 2021-08-18 by macpie

jdgemm, 2021-08-20:
hi @pdvaneijk can you provide more justification for 923 here: https://github.com/helium/miner/pull/991 is seems like there's quite a bit of pushback from community.