helium/router #338: High "hold_time" making it impossible to downlink to devices

nilotnas, 2021-06-10:
I have not been able to downlink to my devices consistently ever since I have started testing with helium. It seems to work on some days, and not work in the others. I have a hotspot sitting in the same apartment as the devices as well as plenty of coverage from neighboors.

Today, after a good morning of testing with positive results, I am no longer able to get downlinks to my devices. Just now I noticed that the hotspot's "hold_time" is incredibly high (>2 seconds), which - as far as I understand this field - means it is impossible to be able to downlink on time to the devices. Please see the example pic below.

[image]

For the sake of the information, I have been running two devices. One of them runs on V1.0.2 of lorawan specification and the other runs on V1.0.3. These devices work consistently with other network servers.

macpie, 2021-06-11:
This is most likely due to an issue with the hotspot. We recently issued a patch to improve `hold_time` I am wondering if that hotspot got it or not. Would you mind checking in next week to see if it has improved or not?

nilotnas, 2021-06-11:
Thanks for the response. I will continue to monitor and update you next week.

If I may add this is the hotspot getting most of the uplinks and it seems to be around the devices, but it is not my own. My hotspot rarely shows up and it's sitting in the same apartment as the devices. Could this mean that the `hold_time` for mine and other hotspots around the devices  is even worse?

macpie, 2021-06-14:
It could be. Is your hotspot synced?

PsRaL, 2021-06-15:
@nilotnas how long is the time between the two uplinks your device is transmitting?
I think i do have the same error here in europe.
I am using a Dragino Lt-22222 as Class A Device with the newest firmware.
When my transmit intervall is below 60 seconds, every downlink is correctly reveived. But if i do increase the interval it cannot receive any downlink. I made a PDF 6 months ago when i first encountered that problem, it is even more complicated because the USB debugging of the device writes that some Downlinks are received in RX2. So either the device has a timing problem or the Helium Hotspot / LNS. I tried two different Hotspots on this. One OG Helium and a RAK.

[Dragino LT-22222.pdf](https://github.com/helium/router/files/6657252/Dragino.LT-22222.pdf)

nilotnas, 2021-06-15:
@PsRaL my transmission period varies a lot throughout the device lifetime, but I have observed the issue with intervals below and above 60s (I think I did 15s, 30s and 5 minutes). Currently my test device is sending every 5 minutes and I'm observing the `hold_time` to be around 1500ms, which theoretically means there could still be time for an RX2 downlink like in your case. Have you been observing your hotspots `hold_time`? Also, have you tested in any other LNS?

Adding here that I believe that issue #339 should be related to the topic being discussed here and could be a potential fix/workaround to this problem.

PsRaL, 2021-06-17:
The hold time is 1500 ms in the average as well. But it seems that it varies very much. The lowest i´ve seen is roughly 750ms and the highest is 5000ms.
If the Helium LNS is transmitting the downlink for RX2 shouldnt it be displayed in console? My Downlinks are the same frequency as Uplinks. This is correct for RX1 but RX2 is happening on a fixed frequency or?
I´ll try to set the RX_Delay higher and see what happens.
Best regards.

nilotnas, 2021-06-17:
@PsRaL the RX2 will happen on a fixed frequency and datarate. If you are using a lorawan device compliant to v1.0.2 of the regional parameters, that will always happen on 869.525 MHz using DR0 (SF12BW125).

With regards to RX Delay, the network is the one that issues the command to change it so that both are in sync about when the donwlink will be sent (see `RXTimingSetupReq` in the lorawan spec). Changing it on your device only will make it so you don't receive downlinks at all.

nilotnas, 2021-06-18:
@macpie I have been monitoring my devices for over a week now. Looks like the `hold_time` is varying a lot during the day, but for the most part of it it is above 1500ms (going up to 4500ms at a time). I have at least 5 hotspots in range, so I believe this is a generalized issue.

Should we expect any improvements on tha realm? Thanks.

nilotnas, 2021-06-18:
> It could be. Is your hotspot synced?

Yes, my hotspot and all the hotspots around me that are picking up messages show as synced.

macpie, 2021-06-18:
OK  we should probably do a debugging session, I used to see that issue on my hotspots but since the patch I do not see late packets anymore and my hold time are usually under 200ms. Could you name (or PM me the names on discord) some of the hotspots that have high hold times?

nilotnas, 2021-06-21:
Here are some of the hotspots picking up my uplinks:
`broad-bamboo-elk`
`cheerful-basil-nuthatch`
`sneaky-tiger-elk`
`sneaky-spruce-stallion`
`dazzling-brick-poodle`
`ambitious-juniper-gorilla`

All of them have above 1.5s `hold_time` and up to 4s. These are not hotspots "winning" the packets and not showing as late.

macpie, 2021-06-22:
This also probably relates to late packets, I will be looking into this

Oliv4945, 2021-06-25:
Dear,

I see two potential issues here:

[An issue](https://github.com/helium/router/issues/327) fixed 3 weeks ago prevented all downlinks to be received on RX2. All the devices **must** rejoin the network since the fix have been put in production to benefit from it.

hold_time: From what I understand of [this line](https://github.com/helium/blockchain-core/blob/master/src/state_channel/blockchain_state_channels_client.erl#L618) the `hold_time` is the time between the actual LoRa reception by the gateway and the reception by the Network Server. If it is the case, a latency of 2-4 seconds is huge, and we can guess that the round time trip (uplink received by the gateway->NS->downlink transmitted by the gateway) will be more.
Currently Helium NS sets the `RxDelay` to 1, it means that the device is opening its RX windows 1 and 2 seconds after transmission. So if the round time trip is more than 1 second, RX1 can not be reached. More than 2 seconds RX2 can not be reached.
Some LoRaWAN networks now this RxDelay to 5 (or more) so RX1, RX2 windows are 5 and 6 seconds after TX, in order to cope with this latency issue. So you have statistics on `hold_time` for Helium network?

nilotnas, 2021-06-25:
hi @Oliv4945,

I had the devices join the network the day I opened this issue. As far as I was aware, the RX2 downlink issue had been solved in production already. Furthermore, I agree with your comments on the `hold_time` and `RxDelay`. I do believe that changing the `RxDelay` to a higher value is a good workaround until this `hold_time` thing is figured out.

I don't have good statistics on the `hold_time`  as I am not setup to monitor it continuosly (but maybe I should, to get better data to back my claim).  As of this morning, I still see the `hold_time` having significant variations from ~1000ms up to ~5400ms for the hotspot that has been forwarding the uplink. Attaching 2 examples from this morning below.

[image]
[image]

pdvaneijk, 2021-06-25:
Hi @nilotnas,

I have been checking my hold times and I am seeing values as low as 29ms and as high as 458ms, no where near what you are are seeing. 

{
  "body": {
    "app_eui": "0016C001FFFE0001",
    "dev_eui": "0016C001F0005D77",
    "devaddr": "0D080048",
    "downlink_url": "https://staging-console.helium.wtf/api/v1/down/baad8cdb-d55d-4fc1-a13f-80a594405717/agYBORndmJO2nw3pUqK8F8PDrv0gUf6o/a7609123-e6c9-4543-b82a-c29a5b3ba74a",
    "fcnt": 65,
    "hotspots": [
      {
        "channel": 11,
        "frequency": 904.5,
        "hold_time": 29,
        "id": "1126UtTc11G9nfAv5cuNuqf6DHP1brtGwb4qyF7phGcgsa295Jej",
        "lat": 37.271485949586555,
        "long": -121.94093416985864,
        "name": "nice-clay-pony",
        "reported_at": 1624641447754,
        "rssi": -110,
        "snr": -7.5,
        "spreading": "SF7BW125",
        "status": "success"
      }
    ],

So probably the above gateway is much closer to the physical server that runs console than the gateways that are receiving your data ?

nilotnas, 2021-06-25:
Thanks for the comparison data @pdvaneijk. This makes this issue even more interesting. Even though I would expect a higher `hold_time` due to distance, 2000ms or even 5000ms seems to be a lot. All the hotspots mentioned above are in NYC. 

Any chance you could share your hotspot firmware version so I can compare with ours? Thanks.

pdvaneijk, 2021-06-25:
My tracker is connecting to some different random hotspots nearby, so I unfortunately cannot tell you what version FW they are running. Are you running console or the staging console ?

{
  "body": {
    "app_eui": "0016C001FFFE0001",
    "dev_eui": "0016C001F0005D77",
    "devaddr": "0D080048",
    "downlink_url": "https://staging-console.helium.wtf/api/v1/down/baad8cdb-d55d-4fc1-a13f-80a594405717/agYBORndmJO2nw3pUqK8F8PDrv0gUf6o/a7609123-e6c9-4543-b82a-c29a5b3ba74a",
    "fcnt": 67,
    "hotspots": [
      {
        "channel": 11,
        "frequency": 904.5,
        "hold_time": 141,
        "id": "112wQk6zy8X9vUyq6Xbxpk96gPnV4DW8TRejWnz3HY4iREGCRorA",
        "lat": 37.269607067227376,
        "long": -121.96357606577197,
        "name": "spicy-navy-capybara",
        "reported_at": 1624643127666,
        "rssi": -97,
        "snr": 6,
        "spreading": "SF10BW125",
        "status": "success"
      }
    ],

nilotnas, 2021-06-25:
@pdvaneijk we are running on the console.

pdvaneijk, 2021-06-25:
@nilotnas: I suggest you connected at least one of your devices to the staging console. They have been fixing issues there and testing them out before fixing their production version console:

https://staging-console.helium.wtf

nilotnas, 2021-06-28:
@pdvaneijk @macpie  I just did a little exercise to try out the staging console and it indeed shows a very different `hold_time` and downlinks work just fine. For the same hotspot I get 200ms of `hold_time` in staging and 5400ms in production. This data is from uplinks roughly 6 min apart that hit the same hostpot (see images below).  I will continue to monitor the devices and see if this behavior is consistent.

Staging Console:
[image]

Production Console:
[image]

PsRaL, 2021-06-29:
I did migrate my device also to the staging console.
The hold_time is indeed way better. Mine tend to be 300ms.
With a low transmit interval (<240 seconds) it stays like this.
It is also constant while i increase the transmit interval of my device (11 minutes).
But when i am queuing a downlink and have set my device to a longer transmit interval it jumps up to 1700ms and the downlinks are not correctly executed.
Nevertheless I had one outlier of a repeated downlink with 300ms of hold time. I was not able to recreate it.
For the downlinks i am either using the Console Websocket or Postman. On Production Console i also used Datacake.

Is Helium at all using the RX2 Window? 1700ms is still better than the results of the production Console and should allow for a usage of RX2.

Best regards.

nilotnas, 2021-07-22:
I have done better analysis on the data available from both the production and staging console. Looks like there has been no improvement in production with regards to the `hold_time`. Staging is fine, but still has some high peaks. I am attaching the charts and the data for reference.  Threshold is set at 2 seconds, but we all know that hold_time needs to be well below for downlinks to arrive in time.

Based on the data below, can I get some insight on what is causing this and if there is a path to fixing the high `hold_time`? Should I expect this to improve when console 2.0 is released into production? I

If it cannot be fixed, I belive there is a path to increasing the RxDelay as I previously mentioned. 

Production:
[image]

Staging:
[image]

Source:
[helium-staging-hold-time.csv](https://github.com/helium/router/files/6862799/helium-staging-hold-time.csv)
[helium-prod-hold-time.csv](https://github.com/helium/router/files/6862800/helium-prod-hold-time.csv)
[helium-console-json-logs.zip](https://github.com/helium/router/files/6862807/helium-console-json-logs.zip)

jdgemm, 2021-07-27:
we've pushed out a number of fixes, and optimizations are you still seeing this?

PsRaL, 2021-08-02:
I just moved my device back to the production console and i am still seeing this issue.
Normal hold_time ist between 400 to 500ms.
If a Downlink is queued it jumps up to 1300ms (then a downlink is succesful) or 1950ms (downlink unsuccesful).
Tested it only 8 times and only received 2 Downlinks correctly.

nilotnas, 2021-08-02:
@macpie @jdgemm I have been monitoring my devices ever since the console 2.0 release and I have seen significant improvements on the issue. I still see some high `hold_times`, but it seems those are outliers now rather than normal behavior. Monitoring my devices for another week before I close the issue. Thanks for all the work you put in!

Below is an image showing the improvement over the last 480 uplinks sent by my device.
[image]

jdgemm, 2021-08-02:
Thanks for the update, we are providing another release which will continue to optimize the loading issues and bring performance of production more in line what users were reporting on staging.

jdgemm, 2021-08-06:
Closing this as we believe will be further addressed with miner release here: https://engineering.helium.com/2021/08/06/blockchain-release-snapshot-update.html

closed 2021-08-06 by jdgemm