helium/router #339: Set RxDelay for devices in console

jerryyip, 2021-06-15:
As hotspots get downlink packets from the p2p network, the network latency is higher than the traditional UDP network, which will frequently cause `Packet REJECTED` in pktfwd. To avoid this situation, users may need to set RxDelay from 1 to 5 on their devices.

jdgemm, 2021-08-12:
This has been addressed with recent updates. If you are still seeing please raise an additional ticket.

closed 2021-08-12 by jdgemm

buzzware, 2021-11-25:
How was this addressed? I don't see an RxDelay option in console, and we are seeing downlink issues again (#519, #517), which this would help resolve/avoid. I think this should be re-opened. @jdgemm

jdgemm, 2021-12-01:
new issue https://github.com/helium/router/issues/531 filed here @buzzware