helium/router #343: Multi packets not working - late packets

slavendam, 2021-06-18:
Hi,
there is problem with getting just one hotspot in Integration message all the time.
What I need is to get all hotspots when I do mapping and testing.

This is really important feature, because lot of people try to do mapping and testing, and they can do that just for 1 random hotspot, not with all.

When can we expect to get this fixed?

Thanks :)

macpie, 2021-06-18:
Did you turn or the multi packets feature? By default it will only purchase 1 copy of the packet.

If you did, do you see a lot of late packets? (they are filtered out by default you will need to check Show Dropped Uplinks:Late)

slavendam, 2021-06-19:
Yes, multi packet for one device set to 4 packets, and on another is set to All packets - no difference.
I can see them in Debug mode, and also when check "Show Dropped Uplinks: Late".
This is really important feature!

[image]

jdgemm, 2021-06-19:
Have you confirmed there's other Hotspots in the area to provide additional packets?

illperipherals, 2021-06-19:
I can confirm this.

slavendam, 2021-06-21:
> Have you confirmed there's other Hotspots in the area to provide additional packets?

Yes, there are many hotspots. Same issue have few people from USA and UK.

macpie, 2021-06-22:
Issue confirmed will be looking into it

macpie, 2021-08-12:
This was addressed in https://github.com/helium/blockchain-core/pull/908

closed 2021-08-12 by macpie

slavendam, 2021-08-18:
This issue is still there. 
It is better then before, but still there.  As you can see, we can get 2-3 hotspots sometimes, but there are bunch of these which are dropped.
Same situation in EU and in US. @macpie 

[image]

macpie, 2021-08-18:
@slavendam this is being worked on in here https://github.com/helium/router/issues/396