helium/router #347: 2 values sent at same time

simonsaysjake, 2021-06-23:
Following the system wide outage today, upon coming online, device sent 2 pings at exact same time - aka the router had them in a queue then sent them at the same time. This means 2 values sent to our platform at the same time. The timestamps therefore are not true. Timestamps must remain true for Helium to be trusted with data.
[image]

macpie, 2021-06-23:
Router does not queue uplinks, the hotspots do (when it cannot reach Router). For trust reasons, we cannot mark the timestamp on the hotspot so the timestamp is marked upon reception in Router. This is why, you see this behavior of 2 packets coming one after the other with very close timestamps.

To avoid this potential issue we added the field `hold_time` [see here](https://github.com/helium/router#integrations-data). This field will tell you how long the hotspot was holding the uplink so that users can make a decision on whether to keep the data that could still be valuable or discard it.

jdgemm, 2021-06-23:
Here's more information about `hold_time` [here](https://github.com/helium/router/issues/347#issuecomment-867191978). Also, the way to ensure you're getting the most accurate sensor timestamps is to have the sensors themselves send timestamp information.

closed 2021-06-24 by macpie