helium/router #363: Joining issue on the network - Use of RX2 Windows intead of RX1 for EU868

disk91, 2021-07-07:
Currently we are experiencing a lot of difficulties to join the network, different issues has been identify and impact the router logic:
- Currently, the RX1 windows seems to be the default choice whatever is the signal RSSI. RX1 in EU868 is using 868MHz frequencies limited to 14dBm power. Due to noise around the device and device sensitivity this RX windows is not viable when the hotspot is too far.
- The choice of the RX window should take in consideration the RSSI with a suffisaient margin for giving a priority to RX2.

Due to the duty cycle regulation in Europe 1% on 868 eq 0.3% per channel you can use to Join, the use of RX1 as a préféred choice also makes hotspot transmission busy and to cancel transmissions.

I think RX2 should be the default for join accept and RX1 use in limited cases like RSSI higher than -85dBm to address device in the 1km around (according to https://www.researchgate.net/publication/325176736_Path_Loss_Models_for_Low-Power_Wide-Area_Networks_Experimental_Results_using_LoRa)

This RSSI switch could be a parameter we could configure per router instance.

michaeldjeffrey, 2021-08-02:
Router sends an rx2 window with join-accepts. I don't know if/how it gets used by miners
https://github.com/helium/router/blob/master/src/device/router_device_worker.erl#L900-L908

JayKickliter, 2021-08-11:
> I think RX2 should be the default for join accept and RX1 use in limited cases like RSSI higher than -85dBm to address device in the 1km around

that would be an interesting thing to try on a limited number of hotspots

macpie, 2021-08-12:
@disk91 has this been fixed by https://github.com/helium/router/pull/355 ?

closed 2021-08-23 by macpie