helium/router #376: Staging - Joining and uplinks

Alanhall007, 2021-07-16:
1. Using Helium Console 
2. https://staging-console.helium.wtf/
3. Organiszation - OaI1Z-1Dso08h4HQ1xZ_DUU3FeEryjGT
    Device ID1 ea5dd852-d775-4ec6-8909-edb27264b48d
    Device ID2 1f8d3263-b7fb-4c99-b91f-a2d426a88b1d

Hi, I seem to be getting issues when I try to test the tracking devices slightly outside my own hotspots coverage area - the devices seem to lose connection and fail to join in areas which should have coverage - hopefully something simple.  The devices seem to communicate with hotspots I have witnessed fine but only rejoin and begin to uplink again after they have connected with my own hotspot?  Any ideas?

jdgemm, 2021-07-27:
hi, sorry for the late reply, but is this still an issue?

Alanhall007, 2021-07-28:
Hi,

Yes, still having trouble - I’ll try to remove the test devices and start them again but they weren’t even joining near the home hotspot yesterday.

Thanks
Alan

jdgemm, 2021-08-12:
With the recent update here: https://engineering.helium.com/2021/08/06/blockchain-release-snapshot-update.html did this resolve your issue?

closed 2021-08-23 by macpie