helium/router #387: MQTT JR Information

lyanchev, 2021-07-28:
1. Are you using Helium hosted Console or running your own open source Console?
Helium hosted

2. If you are on Helium hosted Console, which url are you currently on? 
N/A

3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).
N/A

Please describe your issue:
Background: I have an IoT company - we develop software solutions and hardware products based on LoRaWAN and Wi-Fi. We've been in this space for 7 years now.

When creating our integration with Helium (using the console 2.0) we see that we receive only the uplinks. However, it's **extremely** useful to also receive the join requests info. It has many applications - see if the device has been restarted or whether it "lost" connection when using confirmed downlinks or it has some connection problems, etc... And the most important of all for our specific use-case - estimating the battery life of the device, as the join procedure costs significant energy.

With the other integrations we've completed - TTI, Tektelic, Actility, Kerlink (Wanesy), Chirpstack, etc., we receive info about the JR/JA.

jdgemm, 2021-08-12:
Thanks for the request, we've added this to our backlog for development.

macpie, 2021-08-24:
@vicmgs This is a request to send the Join Requests to integrations. I think for compatibility's sake we will need some flag on the UI (disabled by default) to turn this on/off.

vicmgs, 2021-08-24:
is this ready on router side already? let me know so we can coordinate

macpie, 2021-08-24:
> is this ready on router side already? let me know so we can coordinate

It is being worked on, will update when I got something more concrete.

vicmgs, 2021-08-25:
router will receive channel with "receive_joins" attribute in the future from console. Let me know if you want me to change it

michaeldjeffrey, 2021-11-04:
This was completed with #455

closed 2021-11-04 by michaeldjeffrey