helium/router #393: lorawan modules Network access problems

weirongbo, 2021-08-06:
1. MQTT server
2.Helium Network server
3.Gateway: Bobcat 300 Lorawan Otaa
4.Module: RAK5205, lora module with GPS
The problem phenomena are:
At site A: All configurations of background and modules have been completed according to the official operation manual. After at+ JOIN, data can be seen in the background.But for some reason, suddenly at+ Jion couldn't get into the net.I've been working on it for a long time.Later, the device was changed to location B. After normal configuration, the network was successfully connected.We then use the at+send command to send the number and see that the last data on the MQTT is around 7:30 this morning.At 7:40, WHEN I got to the company, I disconnected the power of the gateway, and then sent at+send to the tool, which reported error:99, and then at+ Jion prompt failed to join the network, until now, it is unable to join the network.At +version and at+get_config=device:status are normal. At +get_config= LORa :channel You can see that channels starting with more than 80 are ON

jdgemm, 2021-08-12:
We're having trouble understanding the issue, can you try to restate the problem.

weirongbo, 2021-08-13:
Later, the hotspot and RAK devices were rebooted. After performing all configurations once, they were successfully added

closed 2021-08-13 by macpie