helium/router #395: The issue of rewards for hot spots when they complete data transfer

weirongbo, 2021-08-09:
My hot spots:Alert Sepia Dragonfly has transmitted 585 data packets to the user's MQTT server for a user's device RAK2(ID: 8fafba53-26F4-4b77-A94a-4A12A56D25D5), cost the user 1173 credits,And the MQTT server successfully viewed all the data, everything is normal.But my hotspot doesn't get a little data transfer reward for that?Can you tell me why?

jdgemm, 2021-08-12:
Thanks for raising we're investigating.

macpie, 2021-08-13:
I have investigate these 2 hotspots: [Passive Cotton Salmon](https://explorer.helium.com/hotspots/11em8d5rr2qYxqAyqXLHkvz8wpqBG6VrDsYBvv8XFwmvTpWJDx7/activity) and [Alert Sepia Dragonfly](https://explorer.helium.com/hotspots/11uBQAviVrbqaVbm3EiMqe5eMwLwCQYZEd26mZ1kVpep4JrvYcz/activity). Both are getting rewards due to transferring data (see screenshot bellow or link above).

One thing to know about data rewards is that they are not distributed right away and could be delayed as they are only calculated once the state channel is closed. For more information on data transfer rewards please see [documentation here](https://docs.helium.com/blockchain/mining/#hip10-and-variable-hnt-rewards-for-network-data-transfer) and more in depth [code here](https://github.com/helium/blockchain-core/blob/master/src/transactions/v2/blockchain_txn_rewards_v2.erl#L1254).

[image]
[image]

closed 2021-08-13 by macpie