helium/router #396: Multi packet issues

BFGNeil, 2021-08-10:
Running the Wio Terminal mapper and I have all packets turned on.

Lots are returning late, even tho its by MS, and often means I get a lost response (even tho one went through)

example here:

[image]

jdgemm, 2021-08-12:
We're investigating.

macpie, 2021-08-13:
@BFGNeil Could you find out what was the hold time for the initial and late packet? The `hold_time` is a value given to the integrations, you can also use the debug mode to find it.

BFGNeil, 2021-08-14:
All I can see is: 

{
  "id": "3504a146-a708-4686-94e8-847c6f976060",
  "router_uuid": "1c466a1b-2cd3-4ac1-a23a-ebfce68f672f",
  "category": "uplink_dropped",
  "sub_category": "uplink_dropped_late",
  "description": "Late packet",
  "reported_at": "1628931730065"
}

am I missing something? debug on console doesn't seem to have hold_time

BFGNeil, 2021-08-14:
"hold_time": 475, late packet does not list a hold_time

macpie, 2021-08-16:
Looks like the hotspot held on to the packet a little longer than the the threshold (400ms currently). A few fixes were implemented to improve hold time but it does not seems to always be enough.

I believe we should rethink how we timeout for packets and maybe allow longer if we do not intend to send a downlink to the device. We then will be able to capture more packets and avoid the infamous "late packet issue".

This will be my top priority.

macpie, 2021-08-19:
This should be improved with #410. 

Following our release schedule we will tag today for a week of testing and release next week.

macpie, 2021-08-23:
This is not fully fixed just yet

macpie, 2021-08-23:
Moving here https://github.com/helium/router/issues/412

closed 2021-08-23 by macpie