helium/router #402: router sc list command lacking documentation

mfalkvidd, 2021-08-16:
1. Are you using Helium hosted Console or running your own open source Console?

My own

2. If you are on Helium hosted Console, which url are you currently on? 

Not applicable

3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).

Not applicable

Helium Console version: Built from d2b6af6bb85acda507cc4cfbcc83a8d2dc786418

Please describe your issue:

When executing `docker exec helium_router router sc list` I get a table that looks like this:
```
+-----------+---+----+-----+-----+----+---+----+------+-------+---------+--------------------------+
|    id     |non|stat|is_ac|expir|expi|amo|num_|num_pa|partici|max_parti|        root_hash         |
+-----------+---+----+-----+-----+----+---+----+------+-------+---------+--------------------------+
|"hidden-bei|126|"clo|true |96595|true|100| 64 |  63  |   1   |  1500   |"1fjCaWZfRo49GYNj4Q9vRbnW/|
+-----------+---+----+-----+-----+----+---+----+------+-------+---------+--------------------------+
```

1. What do the different fields mean?
2. How do I make the table wider so that the information is not truncated? My terminal is much wider than the rendered table.

mfalkvidd, 2021-08-17:
Same questions apply to `docker exec helium_router router device all`. Let me know if you want a separate issue for that.

macpie, 2021-08-18:
1. I will add some documentation for this CLI command.
2. The table size depends on your screen size I believe mine is not truncated.

```
+-------------------------+-----+------+---------+---------+-------+------+-------+-----------+------------+----------------+----------------------------------------------+
|           id            |nonce|state |is_active|expire_at|expired|amount|num_dcs|num_packets|participants|max_participants|                  root_hash                   |
+-------------------------+-----+------+---------+---------+-------+------+-------+-----------+------------+----------------+----------------------------------------------+
|"curved-butter-dragonfly"|  0  |"open"|  true   | 971838  | false |10000 |   0   |     0     |     0      |      1500      |                      []                      |
| "quaint-heather-jaguar" |  6  |"open"|  true   | 971821  | false |10000 |   8   |     4     |     2      |      1500      |"2UQSeyXwnvrzQz7H5CVgEQySuQxYyXHA2qE3wd1t5U8="|
| "thankful-punch-canary" |  0  |"open"|  false  | 972574  | false |10000 |   0   |     0     |     0      |      1500      |                      []                      |
+-------------------------+-----+------+---------+---------+-------+------+-------+-----------+------------+----------------+----------------------------------------------+
```

mfalkvidd, 2021-08-18:
@macpie My terminal is much wider than the rendered table. I'll post a screen shot tomorrow.

mfalkvidd, 2021-08-18:
Could be an issue with how I am executing `docker exec`. Maybe I need to add the `-t` parameter to allocate a tty, so the environment knows how wide the terminal is. I can verify this tomorrow and report back.

macpie, 2021-08-18:
> Could be an issue with how I am executing `docker exec`. Maybe I need to add the `-t` parameter to allocate a tty, so the environment knows how wide the terminal is. I can verify this tomorrow and report back.

Yes indeed, good catch
```
$ docker exec -it helium_router _build/default/rel/router/bin/router sc list
+-------------------------+-----+------+---------+---------+-------+------+-------+-----------+------------+----------------+----------------------------------------------+
|           id            |nonce|state |is_active|expire_at|expired|amount|num_dcs|num_packets|participants|max_participants|                  root_hash                   |
+-------------------------+-----+------+---------+---------+-------+------+-------+-----------+------------+----------------+----------------------------------------------+
|"curved-butter-dragonfly"|  0  |"open"|  true   | 971838  | false |10000 |   0   |     0     |     0      |      1500      |                      []                      |
| "quaint-heather-jaguar" |  6  |"open"|  true   | 971821  | false |10000 |   8   |     4     |     2      |      1500      |"2UQSeyXwnvrzQz7H5CVgEQySuQxYyXHA2qE3wd1t5U8="|
| "thankful-punch-canary" |  0  |"open"|  false  | 972574  | false |10000 |   0   |     0     |     0      |      1500      |                      []                      |
+-------------------------+-----+------+---------+---------+-------+------+-------+-----------+------------+----------------+----------------------------------------------+

$ docker exec  helium_router _build/default/rel/router/bin/router sc list
+--------------+---+---+-----+-----+----+---+----+------+-------+---------+-------------------------+
|      id      |non|sta|is_ac|expir|expi|amo|num_|num_pa|partici|max_parti|        root_hash        |
+--------------+---+---+-----+-----+----+---+----+------+-------+---------+-------------------------+
|"curved-butter| 0 |"op|true |97183|fals|100| 0  |  0   |   0   |  1500   |           []            |
|"quaint-heathe| 6 |"op|true |97182|fals|100| 8  |  4   |   2   |  1500   |"2UQSeyXwnvrzQz7H5CVgEQyS|
|"thankful-punc| 0 |"op|false|97257|fals|100| 0  |  0   |   0   |  1500   |           []            |
+--------------+---+---+-----+-----+----+---+----+------+-------+---------+-------------------------+

```

mfalkvidd, 2021-08-18:
Great, thanks for verifying. When the full names of the columns are visible, the names are pretty self-explanatory. I'm ok with closing this without additional documentation.

macpie, 2021-08-18:
Sounds good to me!

closed 2021-08-18 by macpie