helium/router #404: docker exec helium_router router versions results in stack trace

mfalkvidd, 2021-08-17:
`docker exec helium_router router versions`
```
escript: exception error: bad generator {badrpc,
                                {'EXIT',
                                 {undef,
                                  [{release_handler,which_releases,[],[]},
                                   {rpc,'-handle_call_call/6-fun-0-',5,
                                    [{file,"rpc.erl"},{line,197}]}]}}}
```
`docker exec helium_router router versions --help`
```
escript: exception error: {function_clause,
                     [{local,versions,
                          [{router,"-name",'router@127.0.0.1',router},
                           [{version,"--help"}]]}]}
  in function  escript:code_handler/4 (escript.erl, line 854)
  in call from erl_eval:local_func/6 (erl_eval.erl, line 567)
  in call from escript:interpret/4 (escript.erl, line 788)
  in call from escript:start/1 (escript.erl, line 277)
  in call from init:start_em/1
  in call from init:do_boot/3
```

I expected to see which version I am running

macpie, 2021-08-18:
This would be the release version from the Erlang release, we do not use this.

We are using docker for versioning at the moment. `docker ps` will print the running container and the image used, you can then check on our [image repo](https://quay.io/repository/team-helium/router) (each image is tagged with same "tag" in github)

closed 2021-08-18 by macpie

mfalkvidd, 2021-08-18:
@macpie perhaps the broken command should be removed then, to avoid leading more people astray?

docker ps says ”latest” which I guess won’t be very useful?

macpie, 2021-08-18:
I do think we can remove this specific command without removing more useful ones unfortunately.

Latest will point to a tag on quay https://quay.io/repository/team-helium/router currently `2021.08.10.0`

mfalkvidd, 2021-08-18:
Latest will point to the tag that was latest when the docker was last updated. That might not be the currently lastest on Quay.

But I think I'll be able to approximate the version based on what date the docker was updated.

macpie, 2021-08-18:
This will give you the details `docket image  inspect quay.io/team-helium/router:latest`