helium/router #405: Unable to refill DC balance on self-hosted router

mfalkvidd, 2021-08-17:
I went to https://my_domain/api/organizations/ to retrieve an organization ID (since I can't get any info from ` router dc_tracker info all` due to https://github.com/helium/router/issues/400 )
I then executed the following command:
`docker exec helium_router router dc_tracker refill d0bd698c-b0d8-4168-984a-8ce431549901 -b 50 -n 1 --dry-run`
and got the result
```
Could not find the Organization you're looking for
Create <<"d0bd698c-b0d8-4168-984a-8ce431549901">> by passing the --force flag

ok
```
I expected to be able to refill the organization. Should I use a different ID? If so, how do I get it?
`docker logs helium_router -f` does not display any information when the command is executed.

macpie, 2021-08-18:
I think I answered this in https://github.com/helium/router/issues/400

If the org is not in cache yet you can use `--force` to created it:

```
$  router dc_tracker refill test -b 20 -n 1
Could not find the Organization you're looking for
Create <<"test">> by passing the --force flag
ok

$ router dc_tracker refill test -b 20 -n 1 --force
Created <<"test">> with 20 @ epoch 1
ok

$ router dc_tracker info all
 test --> [balance: 20] [nonce: 1]
```

I need to 2x check the impact on console to override/refill the balance on Router (as it only act as a cache)

macpie, 2021-08-18:
After chatting with the team we decided that there is misunderstanding cause by the fact that this CLI is only for router's cache.

We will resolve it by updating the CLI to include data from Console's DB and also make the refill not only a cache refill but also a Console DB update (Console API are already available) so that we avoid any future issues.

macpie, 2021-08-23:
This https://github.com/helium/router/pull/411 should clear up the situation a little, please see readme for more details.

closed 2021-08-23 by macpie