helium/router #407: Self-hosted router repeats error bad argument in call to erlang:list_to_integer([]) router_sc_worker:get_sc_buffer:395

mfalkvidd, 2021-08-17:
My self-hosted router repeats this error every 15 seconds. The router seems to work, but having errors in the log seems liek a bad thing.
```
[2021-08-17 12:04:45.353] <0.27303.0> [error] [router_sc_worker:get_sc_buffer:395] gen_server router_sc_worker terminated with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:04:45.353] <0.27303.0> [error] [router_sc_worker:get_sc_buffer:395] CRASH REPORT Process router_sc_worker with 0 neighbours crashed with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:04:45.353] <0.1477.0> [error] [router_sc_worker:get_sc_buffer:395] Supervisor router_sup had child router_sc_worker started with router_sc_worker:start_link(#{}) at <0.27303.0> exit with reason bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395 in context child_terminated
[2021-08-17 12:05:00.364] <0.27330.0> [error] [router_sc_worker:get_sc_buffer:395] gen_server router_sc_worker terminated with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:05:00.364] <0.27330.0> [error] [router_sc_worker:get_sc_buffer:395] CRASH REPORT Process router_sc_worker with 0 neighbours crashed with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:05:00.365] <0.1477.0> [error] [router_sc_worker:get_sc_buffer:395] Supervisor router_sup had child router_sc_worker started with router_sc_worker:start_link(#{}) at <0.27330.0> exit with reason bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395 in context child_terminated
[2021-08-17 12:05:15.373] <0.27350.0> [error] [router_sc_worker:get_sc_buffer:395] gen_server router_sc_worker terminated with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:05:15.373] <0.27350.0> [error] [router_sc_worker:get_sc_buffer:395] CRASH REPORT Process router_sc_worker with 0 neighbours crashed with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:05:15.374] <0.1477.0> [error] [router_sc_worker:get_sc_buffer:395] Supervisor router_sup had child router_sc_worker started with router_sc_worker:start_link(#{}) at <0.27350.0> exit with reason bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395 in context child_terminated
[2021-08-17 12:05:30.380] <0.27369.0> [error] [router_sc_worker:get_sc_buffer:395] gen_server router_sc_worker terminated with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:05:30.380] <0.27369.0> [error] [router_sc_worker:get_sc_buffer:395] CRASH REPORT Process router_sc_worker with 0 neighbours crashed with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:05:30.381] <0.1477.0> [error] [router_sc_worker:get_sc_buffer:395] Supervisor router_sup had child router_sc_worker started with router_sc_worker:start_link(#{}) at <0.27369.0> exit with reason bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395 in context child_terminated
[2021-08-17 12:05:45.389] <0.27432.0> [error] [router_sc_worker:get_sc_buffer:395] gen_server router_sc_worker terminated with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:05:45.389] <0.27432.0> [error] [router_sc_worker:get_sc_buffer:395] CRASH REPORT Process router_sc_worker with 0 neighbours crashed with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:05:45.390] <0.1477.0> [error] [router_sc_worker:get_sc_buffer:395] Supervisor router_sup had child router_sc_worker started with router_sc_worker:start_link(#{}) at <0.27432.0> exit with reason bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395 in context child_terminated
[2021-08-17 12:06:00.396] <0.27453.0> [error] [router_sc_worker:get_sc_buffer:395] gen_server router_sc_worker terminated with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:06:00.396] <0.27453.0> [error] [router_sc_worker:get_sc_buffer:395] CRASH REPORT Process router_sc_worker with 0 neighbours crashed with reason: bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395
[2021-08-17 12:06:00.397] <0.1477.0> [error] [router_sc_worker:get_sc_buffer:395] Supervisor router_sup had child router_sc_worker started with router_sc_worker:start_link(#{}) at <0.27453.0> exit with reason bad argument in call to erlang:list_to_integer([]) in router_sc_worker:get_sc_buffer/0 line 395 in context child_terminated
```

macpie, 2021-08-18:
@mfalkvidd thanks for the report. There was a bug when a value was not set.

Also as we add new configuration we usually put it in https://github.com/helium/router/blob/master/.env-template#L21

mfalkvidd, 2021-08-18:
Excellent, thanks. I'll add the value tomorrow and report back.

macpie, 2021-08-18:
Issue fixed  in #409

closed 2021-08-18 by macpie

mfalkvidd, 2021-08-19:
Adding the value got rid of the error messages in router. Thanks!