helium/router #413: Pending XOR / Routing Potential Issue / Devices

NotAnotherDivideByZero, 2021-08-24:
I added a device to private helium console (v2). 
Added the device using:
example - DevEUI: ABCDEFG
example - AppEUI: 999aaa999aaa
example - AppKey: 987654321

The sensor is: [](https://www.elsys.se/shop/product/ers-lite/?v=f003c44deab6) an Elsys ERS-Lite
While we await a HotSpot we are using those in the area for which there are three near by:

1. Decent Tawny Hawk
2. Cold Aqua Pelican
3. Sour Laurel Gorilla

 
We we got date activated to show up and were getting some last updated timestamps but no data and the console front end said there was some integration issues. 

After trying a few things we deleted the device and re-added it with all details being exactly the same. The console front end then just showed 'Pending' Device not yet in XOR filter.  

After leaving this for considerable time we deleted it and added it again but with a new AppEUI and AppKey. DevEUI can not be edited and so remained the same.

Again the same pending issue after some time.

We then assuming it is something to do with our console install added the device to the public helium console but with the same details as we had on the private one.  Hence there are now two consoles with the same DevEUI, AppEUI, AppKey. Neither console got any data. 

**- Out of interest what happens to routing in this scenario, which console would it route too ?** 

A few 100 other variations and attempts were made to connect a device all have failed. 

We tried again by deleting all existence of the device in both consoles and re-adding it with new a AppEUI, AppKey to the private console. This again now just has pending. 

Speaking on the discord channel we were recommend to **ROUTER_XOR_FILTER_WORKER=true**. This was done and the full system restarted. Same pending issue problem but now with an extra one added on launch: 
```

helium_console | Request: POST /api/router/devices/update_in_xor_filter
helium_console | ** (exit) an exception was raised:
helium_console |     ** (RuntimeError) expected action/2 to return a Plug.Conn, all plugs must receive a connection (conn) and return a connection, got: nil
helium_console |         (console) lib/console_web/controllers/router/device_controller.ex:1: ConsoleWeb.Router.DeviceController.phoenix_controller_pipeline/2
helium_console |         (phoenix) lib/phoenix/router.ex:352: Phoenix.Router.__call__/2
helium_console |         (console) lib/console_web/endpoint.ex:1: ConsoleWeb.Endpoint.plug_builder_call/2
helium_console |         (console) lib/console_web/endpoint.ex:1: ConsoleWeb.Endpoint.call/2
helium_console |         (phoenix) lib/phoenix/endpoint/cowboy2_handler.ex:65: Phoenix.Endpoint.Cowboy2Handler.init/4
helium_console |         (cowboy) /app/deps/cowboy/src/cowboy_handler.erl:41: :cowboy_handler.execute/2
helium_console |         (cowboy) /app/deps/cowboy/src/cowboy_stream_h.erl:320: :cowboy_stream_h.execute/3
helium_console |         (cowboy) /app/deps/cowboy/src/cowboy_stream_h.erl:302: :cowboy_stream_h.request_process/3
```

.env-router:
//.env-router
```
ROUTER_SEED_NODES=/ip4/35.166.211.46/tcp/2154,/ip4/44.236.95.167/tcp/2154
ROUTER_CONSOLE_ENDPOINT=http://console:4000
ROUTER_CONSOLE_WS_ENDPOINT=ws://console:4000/socket/router/websocket
ROUTER_CONSOLE_SECRET=SomeSecret1
ROUTER_OUI=10
ROUTER_DEFAULT_DEVADDR=AAQASA==
ROUTER_SC_OPEN_DC_AMOUNT=1000
ROUTER_SC_EXPIRATION_INTERVAL=45
# Max time to wait for uplinks in ms
ROUTER_FRAME_TIMEOUT=300
# Turn on/off xor filter worker (anything else than true is off)
ROUTER_XOR_FILTER_WORKER=true
```
.env
```
SECRET_KEY_BASE=
CLOAK_SECRET_KEY=
GUARDIAN_SECRET_KEY=
ROUTER_SECRETS=
OUI=10
SELF_HOSTED=true
INITIAL_ORG_GIFTED_DC=10000
ENV_DOMAIN=my.domain.net
AUTH_0_CLIENT_ID=aClientID
AUTH_0_DOMAIN=myDomain.eu.auth0.com
AUTH0_BASE_URL=https://myDomain.eu.auth0.com
MAILGUN_API_KEY=MailGunSecretAPIKey
SITE_DOMAIN_MAILGUN=notify.myDomain.net
DATABASE_DB=console
DATABASE_HOST=postgres
DATABASE_USER=postgres
DATABASE_PASSWORD=SomePassword
```

Happy to chat through, share a screen etc.. on discord if this helps debug things quicker.

macpie, 2021-08-24:
Thanks for the detailed report!

`Router` side potential fixes:

1. For the APP/DEV EUI to make into the blockchain the  `ROUTER_XOR_FILTER_WORKER` need to be set to`true` (description might need to be a little more explicit there).
2. Make sure that your Router "account" has DC to pay for the xor filter update transactions (you can follow the xor filter worker  by tailing the logs for `router_xor_filter_worker`). Price will vary on how many devices are added but it is a minimum of 35K DC I believe (just due to transaction fee).
3. The xor filter is on a timer, meaning that it can take up to 10 minutes + blockchain txn time to propagate an update to the network (note that there was a blockchain issue today that might have prevented your transaction to go threw). For testing purposes you can force an update at anytime via the CLI `router filter ...`. I realize now that documentation has not been updated for this section of the CLI (will fix asap).

For the `Console` crashes, I'll refer to @vicmgs 

Please update us if this resolves your issue.

_PS: I believe that we should also add more documentation around xor filter as it is a specificity or our LoraWan Network Server. We will use this issue as  reference to update the docs._

macpie, 2021-08-27:
Yes we should probably do this live, please post in the `#console` channel on Monday I will keep an eye on the chat (I am on PST timezone).

I'll then make sure to post resolution in here for future reference.

NotAnotherDivideByZero, 2021-08-28:
Will do.  I was also chatting to ttague to set up a meeting so you may get asked twice.Chat Monday and thanks again for the help.Kind regards,Richard

michaeldjeffrey, 2021-08-31:
Router was struggling to get a token from Console to start communicating about devices. 
When self-hosting an instance of Console/Router  [.env-template](https://github.com/helium/router/blob/cb5cf1a7c962e90be61fa2fc29d44fe31434b670/.env-template#L24-L25) values here are correct for getting connected. The connection is not going outside the server. Also, the docker ports are not being mapped to OS ports by themselves.

mfalkvidd, 2021-08-31:
@michaeldjeffrey thanks for posting the solution. Just to make things clear, do you mean that `helium_console:4000` in `ROUTER_CONSOLE_ENDPOINT` and `ROUTER_CONSOLE_WS_ENDPOINT` need to stay unmodified from the template file, while `ROUTER_CONSOLE_DOWNLINK_ENDPOINT` requires that `helium_console:4000` is replaced by the external/public url?

from the template:
```
# Console's connection info (see https://github.com/helium/console)
ROUTER_CONSOLE_ENDPOINT=http://helium_console:4000
ROUTER_CONSOLE_WS_ENDPOINT=ws://helium_console:4000/socket/router/websocket
# Public facing endpoint to POST downlinks for devices
ROUTER_CONSOLE_DOWNLINK_ENDPOINT=http://helium_console:4000
```

michaeldjeffrey, 2021-08-31:
@mfalkvidd that is correct.

I'm going to be doing some more research into why external endpoints don't seem to work and I will update the docs with my findings.