helium/router #419: Unknown Decoder

BFGNeil, 2021-08-30:
1. Are you using Helium hosted Console or running your own open source Console?

Helium Hosted Console

2. If you are on Helium hosted Console, which url are you currently on? 

https://console.helium.com/devices/17daa9f2-8dd5-4d63-87cb-a6d44896a315

3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).

Org: BFGNeil
DevID: 17daa9f2-8dd5-4d63-87cb-a6d44896a315

Please describe your issue:

Unknown decoder in debug screen, dropped device, function, integration 2 times now but always returns unknown decoder. 

I can grab the payload and run it through the function and it outputs in the correct format.

macpie, 2021-08-30:
Are you using a custom decoder?

BFGNeil, 2021-08-30:
Yes:

function Decoder( bytes, port ) {
  
  var decoded = {};

  decoded.latitude = convertBin( bytes, 0, 4 ) / 100000.0;
  decoded.longitude = convertBin( bytes, 4, 4 ) / 100000.0;
  decoded.altitude = convertBin( bytes, 8, 2 );
  decoded.speed = convertBin( bytes, 12, 2);
  decoded.hdop = bytes[10];
  decoded.battery = bytes[11] / 10;
  decoded.sats = bytes[14];
  decoded.accuracy = bytes[10];
  
  return( decoded );
}

function convertBin( buffer, offset, len, format="LSB" ){
  var res = 0;
  for( k = offset; k < offset + len; k++ )
      res += buffer[k] << ( 8 * ( (format=="LSB") ? k : len - k ) );  
  return( res );
}

PR #476 (2021-10-30): Allow more whitespace around arg-list in decoders
We don't provide error checking while decoder functions are being
written, and sometimes arguments feel like they're being hugged a bit
too much in their arg-list. Let's allow for some breathing room.

Fixes #419
commit ad36f6b8 (2021-10-30): Allow more whitespace around arg-list in decoders (#476)

We don't provide error checking while decoder functions are being
written, and sometimes arguments feel like they're being hugged a bit
too much in their arg-list. Let's allow for some breathing room.

Fixes #419
closed 2021-10-30 by michaeldjeffrey