helium/router #423: Join-Accept and downlink RX window choice for Europe

disk91, 2021-09-01:
In Europe (and some others) the RX Windows uses different frequencies, TX power rules and Duty cycle rules. Currently the RX window 1 is privilegiated by the hotspot / router and this is not a good choice.

RX window uses the devices frequencies (867-868MHz) with a limited duty cycle 1% and a limited TX power 14dBm. As a consequence, the device's capacity to receives these communications is limited and the loss rate is high. This is due to the asymmetry of the reception capability between a device and a hotspot.

My proposal is to prefer Rx Window 2 for EU868 (and other asymetric area) as soon as the signal from the device is under a given limit:
- Under given RSSI limit - downlink/ack and join accept uses RX2 and 27dBm Tx power on 869MHz
- Above given RSSI limit - downlink/ack and join accpet uses RX1 and 14dBm Tx power on 867-868MHz

I propose to use -80dBm as RSSI Limit as an abitrary choice. I recommend to have a parameter for this.

0dBm to -80dBm is corresponding to short range communication (about 1350m FLS). In this case the 14dBm TX power will reduce the coverage of the hotspot and reduce the global noise on the radio frequency. It will also limit the risk of device receiver saturation.

-80dBm to -140dBm is corresponding to long range communication where the maximum of power is recommended to ensure

The use of RX Windows 2 also use the 10% duty cycle frequency, this is reducing the risk of channel saturation when a large number of downlink communication are proceeded. On top of this it limits the communication on 867-868MHz and reduce the risk of collision with Uplink & PoC comunication.

jdgemm, 2021-09-28:
internal ref: 423

mikev, 2021-09-30:
The proposal is reasonable.  The spec leaves the RX window choice up to the LNS, so we're free to choose either window 1 or 2.

While reviewing the code we also noticed a bug related to which Gateway we choose.  Currently in order to send a reply we require packets from several Gateways.  We then choose the Gateway with the lowest rssi for a response.  Unfortunately we were choosing a random Gateway.  This will be fixed too.