helium/router #427: Documentation Effort

macpie, 2021-09-07:
- [ ] Docs need their own directory
- [ ] CLI
- [ ] Env variables
- [ ] Metrics
- [ ] Lora support

buzzware, 2021-11-12:
https://github.com/helium/router/issues/507

dpezely, 2022-03-23:
Adding to this list:
- [ ] Runbooks -- e.g, what to do at 2 AM when Pager Duty alerts and you're half asleep

In the short term, I'll add instructions based upon extraction of shell history from various production servers with any sensitive information elided, of course.  Most of this *could* go within each repo's `docs` subdirectory: router, packet-purchaser, etc. but open to suggestions.

Examples:
- start/stop runtime via `docker` or `docker-compose`, etc.
- create a snapshot
- load from a snapshot
- interrogate health of the system via *some* of the aliases; e.g., `pp sc list` for summary of State Channels
- etc.

dpezely, 2022-03-24:
Draft PR for new router runbook: https://github.com/helium/router/pull/667

closed 2022-05-23 by macpie