helium/router #430: my iot device not jion

weirongbo, 2021-09-08:
my device id:e53a95e7-dc87-406d-b9a7-903595e70f06
my Organizations:weiorngbo
my hotspots:Shaggy Ruby Guppy   and  Hollow Ivory Python
I executed the following command:
at+set_config=lora:work_mode:0
at+set_config=lora:join_mode:0
at+set_config=lora:class:0
at+set_config=lora:region:470
at+set_config=lora:dev_eui:6081F94B7609A3F4
at+set_config=lora:app_eui:6081F943ADFBB54D
at+set_config=lora:app_key:79EE384F72617A73A68B8664D66585A4
after run   at+join,The following information is displayed on the control software:
OTAA Join Start...
[LoRa]:Join retry Cnt:1
[LoRa]:Join retry Cnt:2
[LoRa]:Join retry Cnt:3
[LoRa]:Join retry Cnt:4
[LoRa]:Join retry Cnt:5
[LoRa]:Join retry Cnt:6
ERROR: 99
Why does this happen?What should I do?

macpie, 2021-09-24:
@weirongbo This is most likely due to the load issue we are experiencing at the moment. Fix is in the work.

_side node: do not post your `app_key` and others as they are your devices credentials (I already removed them)_

macpie, 2021-10-04:
I believe that this should be fix by latest tag `2021.10.04.0`

closed 2021-10-04 by macpie