helium/router #433: LoRaWAN Profile Request: No CFList on EU868

lthiery, 2021-09-16:
There are some LoRaWAN devices being sold for regions like Europe, where they only know how to function on 3 channels. This is technically LoRaWAN compliant and therefore we need a way for the user to disable CFList (the expansion onto 5 channels).

macpie, 2021-09-24:
Is that the option you are talking about?

[image]

macpie, 2021-09-24:
We only support disabling for US915 we would need to add support for EU868

jdgemm, 2021-09-28:
internal ref: 10293

jdgemm, 2021-10-15:
Thanks for raising this issue, we have come up with a plan and have put added it to our backlog.

closed 2021-10-15 by jdgemm