helium/router #435: Can't receive downlink in my device

marciojtjr, 2021-09-18:
I am developing a new device, using a RF95 LoRa module.
I don't have a full LoRaWAN stack and I am testing the integration with Helium Network. I manually configure all the radio parameters and I can successfully perform the OTAA Join procedure.
For testing purposes I am always using the same channel for uplink (`904.60 | SF8BW500`) and downlink (`923.90 | SF7BW500`).
[image]

After the Join process is done, I can successfully send uplink messages and even decode it on my integration, but I can't receive the downlink message in my device.

I've changed my device to start listening immediately after finish sending the uplink and keep listening for over a minute, trying to mitigate any timing issue, but no luck. Sometimes, like in the attached picture, I see a '-1' as the Frame counter for the downlink and I am wondering if this can be related to my problem.

Since I can successfully receive the JoinAccept message, I don't believe it is a device's radio mistake, but without class C support I don't see other possibilities for debugging the downlink message.

marciojtjr, 2021-09-20:
Update:
I don't kow if this is related to my downlink issue, but checking the JSON Debug data I saw many huge values for `hold_time`. 
Sometimes varying from 1200 to 5000 and other times varying from 12500 to 42000.

[image]
----
[image]

marciojtjr, 2021-09-23:
Another information about this issue is that every unconfirmed uplink is throwing a downlink. All downlinks has link_adr_request comands and the FCounter = -1
[image]

It looks like the router is sending an invalid configuration. I'm using US902 and inside link_adr_request we have ChMaskCtl=7 and ChMask=65280. This Channel Mask Control means that all 125kHz channels are disable and the ChMask bits are related to enabling/disabling of channels 64 to 71 (8 channels). Surprisingly, this Channel Mask value enables channels index 9 to 15 !
[image]

lthiery, 2021-09-23:
I believe the -1 is a display bug. 

Insofar as the LinkAdrReq, we are indeed trying to configure channels 9 to 15 (and disable the rest) as those are the channels we operate on. If your stack does not like the LinkAdrReq, you always have the option of NACK'ing it, but our LNS will continue sending the LinkAdrReq until there is an ACK or NACK.

marciojtjr, 2021-09-24:
Ok, I'll keep an eye on the -1. I remember it used to show correct numbering a few weeks ago.

Regarding the LinAdrReq messages I'm not clear. For US902 channels plan, with Channel Mask Control = 7, the channels available would be the 8 500kHz (channels 64-71). This way the router should never try to enable channels 9-15. As per my understandings, the bits to be used on Channel Mask Control shoudl be only 0-8 (mapping to 64-71 directly). Can you please explain what is the meaning of enabling channels bits 8-15?

Also, this is not my main issue. I only added this information here to try to identify the origin of the problem. My main issue is that I can never receive any downlink. I only receive the JoinAccept, which makes me believe I don't have a radio issue on my device.
Is it guaranteed that the downlink was actually sent?

lthiery, 2021-09-24:
Downlink reliability [is currently degraded due to high load](https://discord.com/channels/404106811252408320/830872423675068416/886014673835659285).

Last I heard, we will have something out next week that should improve things.

marciojtjr, 2021-09-25:
I realized that I was misinterpreting the MAC commands.
Now I reanalyze it and understand the first command disables all 125kHz channels and the second enables channels 8-15. Now I see there's nothing wrong with this commands.
[image]

Using one of the recommended 125kHz channels, my results are the same. I mean, I cannot receive a downlink on my device.

closed 2021-10-04 by macpie