helium/router #436: HTTP Integration - unknown hotspot location returns "unknown" where a float is expected

disk91, 2021-09-22:
The fields lat and long for hotspot location can take a double value or a String value with content "unknown"
```
{
        "channel": 7,
        "frequency": 867.9000244140625,
        "hold_time": 0,
        "id": "149j3nj37KLqdxZWY3HbATMRhteGe7g2aYKAQmhHSSwrqWJYcdf",
        "lat": "unknown",
        "long": "unknown",
        "name": "main-amethyst-hamster",
        "reported_at": 1632333026938,
        "rssi": -75,
        "snr": 10,
        "spreading": "SF7BW125",
        "status": "success"
      }
```
This is a problem for a structured developpement language does not accepting multiple format for a signe JSON entry.
I recommend to not provide this field when unknown or to set it up to 0.0 / 0.0 as this possible will not have any real hotspot located here.

macpie, 2021-09-24:
Problem is `0,0` is arguably a location.

disk91, 2021-09-27:
I agree, but "unknown" is not a double and it's an industry standard to report 0,0 when you don't have a position. This is why all platforms are filtering this position. Even if you are really located in (0,0), you generally report (0.0001, 0.003) thanks to the GPS precision so the risk to confuse (0,0) with a real position is really thin.

We could imagine 999.9,999.9 as it is not a valid coordinate. But the side effect could be larger as some implementation will consider values modulo 180/90.

macpie, 2021-10-04:
Fair enough, we will schedule this and communication to community will be important as we are changing data sent to integrations

lthiery, 2021-10-05:
> This is a problem for a structured developpement language does not accepting multiple format for a signe JSON entry.

IMO null or omitting the field altogether is more typical for JSON. Do you know if your system would play well with that?

disk91, 2021-10-05:
Do not having lat and long field could be a solution. Usually we can manage this with default value in the different languages.
In a such choice, make sure the mapper input also supports such situation : currently mapper have a strict syntax incompatible with router output in a such case.

macpie, 2022-05-23:
Considering the time that we  have all been using `undefined` changing it would require versioning payload.

closed 2022-05-23 by macpie