helium/router #443: Router blockchain.db / ledger.db / checkpoints size

disk91, 2021-10-04:
The different .db file of the router are become bigger and bigger and make file system going full
```
4,0K	blockchain
50G	blockchain.db
2,9G	blockchain_swarm
6,9G	checkpoints
14G	ledger.db
25M	log
426M	router.db
422M	state_channels.db
```

Is there a way to limit the size and allow router to run on a reasonable file system size ?

lthiery, 2021-10-04:
IMO, 64GB isn't too bad when logging hundreds of thousands of blocks with hundreds of transactions... 

That being said, you could delete the files and load a snapshot every two to four weeks. That basically allows you to start fresh with only a few hundred blocks and an up to date ledger:

- make a snapshot
- delete all directories except for `blockchain`, where your swarm_key lives
- load the snapshot

With a 64GB system, I think you'll want to do this every two to four weeks. I generally pick a 128GB or 256GB file system so I can worry about it less. Doing this operation will obviously create downtime and I'm not sure how quickly the Router learns about its own state channels and if that starts back up elegantly after deleting the directories.

macpie, 2021-10-04:
We encountered the same issue (more blockchain rather than Router). The team is working on better tools to cleanup old blocks to free some disk space. In the mean time my advice is to increase disk space.

closed 2021-10-04 by macpie

kevinelliott, 2022-05-10:
@lthiery @macpie Has any work been done in the last 6-9 months to cleanup old blocks to free some disk space? I'd like to avoid having to piece together the take_snapshot/delete/load_snapshot flow if possible.

macpie, 2022-05-11:
We are starting work to decouple from the blockchain, see https://github.com/helium/router/issues/690. But it will take sometime. 

> @lthiery @macpie Has any work been done in the last 6-9 months to cleanup old blocks to free some disk space? I'd like to avoid having to piece together the take_snapshot/delete/load_snapshot flow if possible.

We do not have proper documentation for this but Validator does https://docs.helium.com/mine-hnt/validators/troubleshooting#validator-low-disk-space