helium/router #486: update erlang v8

jdgemm, 2021-11-02:
Seeing persistent, recurring issues believe updating to erlang v8 will address.

dpezely, 2021-11-02:
The V8 javascript engine has been upgraded to the latest *stable* release, which is 9.5.172.25.

This involved a minor update to our fork of `erlang_v8`.

The new `v8` binary has been uploaded to the S3 bucket, and we are testing this now along with other router updates.

Vagabond, 2021-11-02:
Note that we need an MacOS build from someone with a fast Mac to put in s3 as well.

someone would checkout 0.5.7 tag of erlang_v8, `make -C c_src build` then check the tests all pass with `make test` and if they're good make a package using `make -C c_src release-tarball` and sending me the `erlang_v8-0.5.7-Darwin-x64.tar.gz` file

cc @michaeldjeffrey

dpezely, 2021-11-03:
Added instructions from the preceding comment to PR 4 of `erlang_v8` repo (our fork).

We should consider x86-64 vs amd64 versions of Macs, but that would be a new feature of `../erlang_v8/c_src/Makefile`.  It looks like there are scripts relying upon python2, which is beyond end-of-life officially, so deeper excavation must be done there.  Maybe address both at the same time?  (Instructions within that PR can serve as a jumping-off point for both aspects.)

dpezely, 2021-12-03:
Although we updated V8 once for this ticket, Google is on a 6 week cadence with releases.  Another `stable` release is available and tested locally to work fine: all our tests passed with the following change to our Makefile:

```makefile
V8_VERSION := "9.6.180.12"
```
However, we're close enough to the *next* release (that will presumably also be tagged `stable` by them), so let's update again then.

closed 2021-12-06 by macpie