helium/router #491: Join Accept Not Receiving By Device

CanEmreUnal, 2021-11-02:
Facing same with [#437](https://github.com/helium/router/issues/478).

Hotspot already updated and working. [2021.10.29.0](https://engineering.helium.com/2021/10/29/miner-emergency-release.html)

Sharing the devices below.

4abb7e0a-aa67-4b28-83ce-44ed5440475b
f515a30a-96f5-41e7-82c7-52e10a617fc2

[image]

CanEmreUnal, 2021-11-08:
Is there any help on here. Thanks

macpie, 2021-11-22:
It seems like `f515a30a-96f5-41e7-82c7-52e10a617fc2` has been sending uplinks recently (2021-11-21 15:49:40).
I have not seen any joins or uplink for `4abb7e0a-aa67-4b28-83ce-44ed5440475b`

closed 2021-11-22 by macpie