helium/router #495: Production Console function decoder output shows error: no_context, status:error

leroyle, 2021-11-04:
1. Are you using Helium hosted Console or running your own open source Console?
  Hosted

2. If you are on Helium hosted Console, which url are you currently on? 
     Production shows the error, Staging does not
3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).
   a0b70b92-1f9a-45ac-98d5-e38711da95ea

Please describe your issue:

- Earlier in the day my device was not able to join the Production console. The device debug view never did show Join or Join Accept. 
- Switched to staging console, connected uplinked as expected
- Back to Production, same "no response" in debug view.
- Reported same to support.

Sometime later my device was able to connect to production console.
I noticed every other "Uplink Integration Request", Decoded Payload was showing
```
"decoded": {
      "error": "no_context",
      "status": "error"
    },
```
- The alternating uplinks were being decoded correctly.
- switched to Staging console, issue was not present, no decoding errors noticed.
- The Device within the console was created pre V2, with no changes since V2
- Region US915, Helium original hotspot,  Texas USA

A sample run follows below.

The current function, also shown below is a copy of my original made this afternoon, but cut down considerably, problem still exhibits in this newer function although perhaps not as consistently alternating between success/failure.

Current Function:
```
// sample 
// Raw:  a740f01528013d0000
// Decoded: 5;1;1;0;15;21:2:61

// base64 hex: d4BQeyACQVQ=
// decoded hex bytes: 7780507b20024154
// Not sure about this decoding as I did // capture the origina: 
// Might be 3:5:2:0:5:123:2:AT
function Decoder(bytes, port) { 
var payload = 
{
"Leroy":"66"
}
payload.by0 = bytes[0]; 
payload.by1 = bytes[1]; 
payload.by2 = bytes[2]; 
payload.by3 = bytes[3]; 
payload.by4 = bytes[4]; 
payload.by5 = bytes[5]; 
payload.by6 = bytes[6]; 
payload.by7 = bytes[7]; 
payload.by8 = 0xff;

return payload;
}
```

Sample run using the above function. This a complete partial run, I've broken up the markup so I could add the GOOD/BAD notices to make it easier to scan.

**GOOD DECODE**
```
Uplink: Integration request Decoded Payload
Event Informationa few seconds ago

{
  "id": "b0464e3e-7528-49dd-863b-805eb42d0d32",
[... 587 lines cut ...]

Hotspot

{
  "channel": 10,
  "frequency": 904.2999877929688,
  "id": "112DnKkS5qVvRoK1TFcbhT7ReezwtNTVmE2zTB2ZnAa3UMZYJ8kN",
  "lat": 32.9550615760182,
  "long": -96.75233778326856,
  "name": "harsh-sepia-gerbil",
  "rssi": -24,
  "snr": 10,
  "spreading": "SF9BW125"
}

console.helium.com
Documentation & Tutorials
How-to Videos
Community Discord
Engineering Blog
Terms & Conditions
© 2021 Helium Systems Inc.
```

.

Max-Plastix, 2021-11-05:
I'm seeing the same Integration Request failure for HTTP Integration as above.
```   "decoded": {
      "error": "no_context",
      "status": "error"
    },```
This happens on about 4 out of 5 Uplinks from Device/Function/Integration that worked previously.  A device that previously Uplinked every minute now shows these correct message timestamps ```
11/5/2021, 2:18:33 PM
11/5/2021, 2:21:31 PM
11/5/2021, 2:22:32 PM
11/5/2021, 2:23:32 PM
11/5/2021, 2:36:32 PM
11/5/2021, 2:38:33 PM
11/5/2021, 2:43:32 PM
11/5/2021, 2:47:32 PM
11/5/2021, 2:52:32 PM
11/5/2021, 2:53:32 PM
```
with the missing entries each reporting an Integration error

The non-HTTP Integrations are still working, and seeing correctly decoded Payloads -- this failure seems specific to the HTTP Integration.  It's common to HTTP Mappers and my own HTTP (Post) integrations, on multiple devices with different Decoders.

The common symptom is that `payload.decoded` does not appear presented in the `Uplink: Integration Request Decoded Payload` for HTTP.
```
{
  "body": {
    "app_eui": "A000000000000101",
    "decoded": {
      "error": "invalid_context",
      "status": "error"
    },
``` while the payload itself is present.  It's almost as if the Decoder fails to execute for some (not all) messages sent to HTTP.

dpezely, 2021-11-11:
Proposed solution:
- Add more fine-grained handling of results from v8
  + e.g., distinguish existing `crash` from available but unprocessed SyntaxError, ReferenceError, etc.
- Apply fixes to `erlang_v8` C++ code to resolve memory leaks
- Validate JavaScript decoder function definitions as at least valid JS as preliminary step to creating a context, so then a bad JS fn doesn't `crash` the v8 port

leroyle, 2021-11-12:
It looks like this has been fixed in staging

closed 2021-12-02 by jdgemm