helium/router #509: CFList switch not working?

Oliv4945, 2021-11-12:
Dear,

I am currently using [Helium hosted console](https://console.helium.com), with device ID `ac26fdce-c1bf-47d5-a844-e0a98e280fed`
The device is using a profile with CFlist deactivated, but my device still gets it. Is this switch working as expected?

Device log. The 0 are channels deactivated by the CFList
```
*************************************
* Receive a Valid downlink RX1 for Hook Id = 2
*************************************

Ch 125kHz
0  0  0  0  0  0  0  0  1  1  1  1  1  1  1  1  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
Ch 500kHz
0  1  0  0  0  0  0  0
DevAddr= 4800036f
MacRx1DataRateOffset= 0
MacRx2DataRate= 8
MacRx1Delay= 1  
```

Console settings - Device
[image]

Console settings - Profile
[image]

michaeldjeffrey, 2021-11-12:
The CFList option you have Disabled in console only applies to putting a CFList in a JoinAccept payload. 
If that option is disabled (as in this case) Router will send a downlink to configure the device for sub-band 2 to match what gateways listen on.

closed 2021-11-12 by michaeldjeffrey

Oliv4945, 2021-11-12:
Hi @michaeldjeffrey, yes this was my understanding. So why the CFList is sent to my device in the joinAccept?