helium/router #510: ADR & manual console downlinks not being received by end device

Net-Metrics, 2021-11-14:
Similar issue to https://github.com/helium/router/issues/497 but still ongoing on my device. I'm in EU863 region, using an original Helium hotspot. I moved my device over to the staging console to test with the updates from issue 497 but still seeing the same behaviour

I have been looking at the timestamps in the console to try and see where the issue is, but unsure if they provide enough info to base this on or not. As far as I understand, device will 'listen' for a downlink 1s & 2s after an uplink. For a join request I can see the accept message is sent 2s after the request

[image]

The accept message sets the Rx windows on the device to 1s

[image]

This is received, and my device joins. With ADR set, my device sends uplinks until #20 is reached, then the console sends a downlink, but it looks as though this occurs only 400mS after the uplink

[image]

My device does not see this, so an endless loop of downlink attempts is started, with nothing being received

[image]

Similar behaviour if I try to send a manual downlink via the console, except it seems to deploy a 500mS delay after the uplink is received

[image]

I am unable to get ADR or a manual downlink to my device, testing with an mClimate Vicki TRV

https://docs.mclimate.eu/mclimate-lorawan-devices/devices/mclimate-vicki-lorawan/vicki-lorawan-device-communication-protocol/communication-concepts

Net-Metrics, 2021-11-23:
Unfortunately I missed the debug logs in the console, but this chart shows one of my sensors that has ADR enabled. I have a log running that records the 'spreading factor' used for uplinks, just so I can keep an eye on battery usage. It was uploading data using SF7BW125, but I assume as it never received a downlink from the console it can be seen stepping all the way up to SF12, with 20 uplinks per step. 

After 24 hours of the console attempting downlinks to set the ADR 1 eventually got through, resetting my sensor to the correct SF7 (it's only about 10m away from my hotspot).

Downlinks **will** eventually get through it seems, but this shows it took approx 100 attempts before a successful one was received (10min uplink period * 24 hours = 96 uplinks)

[image]

mfalkvidd, 2021-11-23:
Could it be that the downlink signal is too strong for the device to receive? Many LoRaWAN users (on Helium and other networks such as TTN) have experiences that short distances between gateway and device cause the RF frontend on the device to be saturated, which results in the device being unable to hear the downlink.

The situation can be compared to someone shouting into your ear using a megaphone from a short distance.

If you can, maybe it is worth trying to move the device a bit further away from the gateway?

nilotnas, 2021-11-25:
@Net-Metrics have a look at issues #338 and #519, as well as the well documented #517.

I'd recommend start looking at the `hold_time` for your uplinks, this is the most likely cause for your downlinks not being received.

Net-Metrics, 2021-11-25:
@nilotnas I read your #519 with interest, but can't see where to check the hold_time, can you point me in the right direction for that so I can compare?

Certainly a few people now having problems with this, pretty unusable for many commercial applications if downlinks cannot be reliably performed

nilotnas, 2021-11-25:
@Net-Metrics for looking at incoming packets I usually open the debug window and tick the "Expand all" box in the top right corner, then do a search for `hold_time`. For looking at more data you can "Export JSON" and search through it.

Net-Metrics, 2021-11-25:
@nilotnas average hold_time for my packets range between 1600-1800, bu have gone as high as 5000+. My device will regularly re-join the network, I assume as it is not receiving any downlinks. When it re-joins (which always seems to be successful on the first attempt), I see the following

```
    "category": "uplink",
    "data": {
      "dc": {
        "balance": 9145,
        "nonce": 1,
        "used": 1
      },
      "devaddr": "0B080048",
      "fcnt": 1,
      "hold_time": 302,
```

Then when uplink 2 is sent it reverts to the following

```
    "category": "uplink",
    "data": {
      "dc": {
        "balance": 9144,
        "nonce": 1,
        "used": 1
      },
      "devaddr": "0B080048",
      "fcnt": 2,
      "hold_time": 1809,
```

So this looks like the same issue as you are seeing, question is why is the `hold_time` low for a `join_request`, but then increases for standard uplinks

buzzware, 2021-12-21:
> This is received, and my device joins. With ADR set, my device sends uplinks until #20 is reached, then the console sends a downlink, but it looks as though this occurs only 400mS after the uplink
> 
> [image]

That 400ms delay is interesting. I'm also seeing it here with downlinks failing https://github.com/helium/router/issues/517#issuecomment-997492361

Net-Metrics, 2021-12-29:
Running on console 2.1.9 now and I had hoped the new `Set Delay Before Rx1 Window` setting in the device profiles would help with my downlink issues, but so far no luck. I have tried setting the Rx1 delay to all 5 settings, forcing a device rejoin after each change but still getting the same cycle of behaviour. `Join request` & `Join accept` always works first time ( with a 400ms `hold_time` give or take. I then get regular uplinks, with `hold_time` averaging between 1900 & 2300.

Device sends 20 uplinks (using SF12BW125), console then tries to send a downlink with the following MAC commands

Data Rate | 5
-- | --
TX Power | 1
Channel Mask | 255
Channel Mask Control | 0
No. of Transmissions | 0

Unfortunately this downlink is never received by my device, it continues to go through the uplink/downlink attempt cycle, until my device gets to approx 96 uplinks (24 hours, 1 * uplink every 15 mins), after which it sends another `join request` and the cycle starts again

I have tried this with all delay settings, and before trying the 2s delay again there, I deleted the device & re-added as a 'new' device to see if that helped, no change in behaviour

[image]

[image]

[image]

buzzware, 2021-12-30:
@Net-Metrics there have been key changes made but not yet released for this. I was told to wait, don't waste your time on it.

mikev, 2022-01-03:
Status Update

1. This issue has priority and is being actively investigated
2. Most likely the root cause from an examination of the logs - the end-device does not physically receive the ADR downlink
3. The root cause issue can be explained by pending rx-delay work.  I'll re-post here when the rx-delay work is deployed.
4. Another potential cause is that rssi from some gateways is too high.  This would result in ADR setting DR and TX too high.  Work is pending  on this issue.
5. ADR is complex - working is pending to add more unit ADR unit tests

mikev, 2022-01-07:
An additional 40 unit tests verifying ADR behavior have been added.  The ADR algorithms, at least internally, is behaving correctly.
https://github.com/helium/router/pull/574

Additional unit tests for edge cases are planned.

mikev, 2022-01-14:
@Net-Metrics - The RX delay should be deployed now.  This must be configured in your device's config profile.  The default remains 1 second.  So you'll need to increase the setting to 5 seconds or more.

buzzware, 2022-01-15:
> @Net-Metrics - The RX delay should be deployed now. This must be configured in your device's config profile. The default remains 1 second. So you'll need to increase the setting to 5 seconds or more.

The UI setting has been on staging for a while, but there was an outstanding issue that caused downlinks to fail on settings not equal to 1 second (the original value). 

Looking at Sprint 68 it doesn't look like this entire feature has been completed or deployed end to end - just trying to save people from confusion.

mikev, 2022-01-28:
The rx delay feature is coming soon and is deployed on Stage now. ETA is Monday, Jan 31st.
#582 (comment)

mikev, 2022-02-14:
The rx delay feature is now deployed to Production. https://github.com/helium/router/issues/597

@Net-Metrics - Can you retest ADR with your end-device? Whether positive or negative, can you post trace logs of the packets?

Net-Metrics, 2022-02-14:
> The rx delay feature is now deployed to Production. #597
> 
> @Net-Metrics - Can you retest ADR with your end-device? Whether positive or negative, can you post trace logs of the packets?

Attached file shows JSON results for a downlink set to default, then 2s, 3s, 4s 5s 6s & 7s.

The most succesfull looking downlink occurred with the default Tx window, but although the downlink was immediately followed by 2 uplinks, the command was not received by my end device. 

[image]

I'm trying to send a downlink to an mClimate Vicki TRV, I'm in UK, EU868. Command I'm sending is 'text' 0E15
[json.txt](https://github.com/helium/router/files/8064270/json.txt)

Net-Metrics, 2022-02-15:
> 

Have I been trying to do downlinks incorrectly? If I want to send the `hex` command `0E10` to my device I was entering the following in the downlink window
[image]
This shows a console calculated Base64 payload of `MEUxMA==`

However, if I use an online Base64 encoder and convert from `0E10` it shows a Base64 payload of `DhA=`

Entering this as my downlink, works fine, downlink received by my device.

[image]

[image]

Have I been doing something wrong with the `Text` downlink option, or is is encoding to Base64 incorrectly?

mfalkvidd, 2022-02-16:
`MEUxMA==` is indeed the **text** `0E10`.

Text ≈ ascii
Text != hex

Net-Metrics, 2022-02-16:
> `MEUxMA==` is indeed the **text** `0E10`.
> 
> Text ≈ ascii Text != hex

I can see that, what I didn't understand was why there is a **text** option. As far as I was aware a downlink would always be either **hex** or **base64**. Is there ever a situation where you would want to download **ascii**?

mfalkvidd, 2022-02-17:
I have seen multiple cases where people send json text strings or plain text to their devices. Doing so is extremely inefficient (in terms of radio spectrum capacity, battery consumption and payload capacity), but people still do it. I can imagine Helium wants to be "n00b-friendly" and therefore have a text option available.

For any professional use, a hex option would make much more sense.

mikev, 2022-02-18:
@Net-Metrics - Examining your logs I noticed that Downlinks are stuck on FCnt 35.  In other words, the Confirmed Downlink is queued but an Ack from the end-device never confirms receipt of the packet.

Were you able to configure the rx_delay feature successfully?

Net-Metrics, 2022-02-18:
Yes, the rx_delay now lets me send downlinks successfully, issue solved as far as I'm concerned.

Thanks for the help

mikev, 2022-02-18:
Excellent!  I'll officially close this issue then.

closed 2022-02-18 by mikev

Net-Metrics, 2022-02-18:
> I'll leaving a closing comment on Text vs Hex vs Base64. There is a misunderstanding in the comments above; data is always sent over the wire via the LoRaWAN protocol as a bit array - https://en.wikipedia.org/wiki/Bit_array <https://en.wikipedia.org/wiki/Bit_array> We don't ever send data as text or base64.
>
> When binary bit data is displayed to a UI it is re-formatted into a more human readable form, e.g. usually hexadecimal, but sometimes base64 encoded. This is just a display format. The underlying bit data is just 1 and 0s.
>
> The console allows input of data as either text or base64 which then gets encoded into bit data. I'm not sure why a Downlink hex option was not provided; that sounds useful. Hope this clarifies matters.

Ideal, thanks. A hex downlink option would be a great idea, nice and simple for a quick modification, and easier for new users to get to grips with