helium/router #519: Unable to donwlink to devices due to high hold_time

nilotnas, 2021-11-24:
This is a follow up on issue #338 .

After a couple of months from confirming that the high `hold_time` issue was fixed I started testing devices again and noticed the issue was back. 

The behavior is the same as I described in issue #338: `hold_time` is above the downlink threshold (which is 2 seconds), and so it will never get picked up by the device. 

I have also noticed a correlation between how often the device transmits data and `hold_time`. When I have a device transmitting every 30 seconds or every minute the `hold_time` goes down and I can see some downlinks going through. For a device transmitting every 5 minutes the `hold_time` goes above the 2-second threshold and it will no longer get downlinks.

I have attached below `hold_time` plots from the past month and from 9 days ago. I am continuously monitoring my devices everyday and have seen no change in this behavior.

[image][image]
[image]

buzzware, 2021-11-25:
Can anyone explain why the transmit period would have anything to do with hold_time?

macpie, 2021-12-18:
Fixed RX delay https://github.com/helium/router/pull/548

closed 2021-12-18 by macpie