helium/router #520: Infinite JoinReq-JoinAcc in Console, device can't join

slavendam, 2021-11-29:
Few of our users reported they can't join network.
They see Join Request and Join Accept in Console, but device is not getting Join Accept.
It is tested with GLAMOS Walker device and RAK WisNode.
Biggest testing was done in Costa Rica where Join Requests were received by 4 different hotspots, but devices can't see none of Join Accepts.

I believe this issue can be connected with long hold time (even console says that Join Accept is sent with 2s delay) or maybe not sending JoinAcc? Same thing was like 2 months ago.

buzzware, 2021-11-29:
Right now on a Heltec, I can't get either of 2 devices to show packets at all in the Helium console.
They have both been working in recent times.
I can see on the Heltec's own lora uplink log that the join requests are received, but nothing in the downlink log.
The network needs better status monitoring.

macpie, 2022-01-04:
Is this still an issue and if yes would you guys mind providing device information (Console device UUID at least)?

slavendam, 2022-01-05:
In last 10 days users were able to join network, not sure what was going on. So I will close this issue now.

closed 2022-01-05 by slavendam