helium/router #522: xor command not accepted by miner

disk91, 2021-11-29:
When using the CLI to force XOR update the CLI manual is printed instead of the expected result
```
docker exec helium_router router device xor 
Usage: 

  device                                        - this message
  device all                                    - All devices in rocksdb
  device --id=<id>                              - Info for a device
  device trace --id=<id> [--stop]               - Tracing device's log
  device trace stop --id=<id>                   - Stop tracing device's log
  device queue --id=<id>                        - Queue of messages for device
  device queue clear --id=<id>                  - Empties the devices queue
  device queue add --id=<id> [see Msg Options]  - Adds Msg to end of device queue

  device prune --commit                         - Removes Devices from Rocks that don't exist in ConsoleMsg Options:

  --id=<id>              - ID of Device, do not wrap, will be converted to binary
  --payload=<content>    - Content for message [default: "Test cli downlink message"]
  --ack                  - Require confirmation from the device [default: false]
  --channel-name=<name>  - Channel name to show up in console [default: "CLI custom channel"]
  --port                 - Port for frame [default: 1]

{error,1}
```

michaeldjeffrey, 2021-11-29:
XOR filter commands have been moved to their own subcommand.

`docker exec helium_router router filter ...`

macpie, 2021-11-29:
We do need to document this

closed 2021-12-13 by macpie