helium/router #524: Improve state channel test suite

jdgemm, 2021-11-29:
Provide a more robust test suite for state channels.

michaeldjeffrey, 2021-12-06:
Can we get a log or something for the case where a packet handler buys an offer, but the internals refuse the offer.

macpie, 2022-01-18:
We need to limit the number of close transaction with conflicts that can be submitted (via chain var). As soon as one conflict is in we should reject anymore close transactions.

Furthermore, when reward time comes, if a state channel close includes any conflict the owner should get "punished" and all of its DC used to open SC should be lost. Note that in this case no hotspot would get rewarded to avoid any attempts at gaming the system

michaeldjeffrey, 2022-02-03:
blockchain-core PR for tracking https://github.com/helium/blockchain-core/pull/1229