helium/router #532: no data transfer and otaa possible since update to 2.1.7

fishpepper, 2021-12-03:
I am using staging-console.helium.wtf to relay data from commercial lora sensors.
Around 1h after the update my data transfers stopped. I can also no longer onboard (OTAA) new sensors on staging-console.helium.wtf.
However doing the same on console.helium.com works fine. So i think it has to do with last nights update... Is it just me or is something broken?

fishpepper, 2021-12-04:
the fix brought it back to life. however it looks like staging is dropping packets again, for some devices I see gaps in the packet nonce and some sensors stopped completely :(

macpie, 2021-12-06:
> The fix brought it back to life. thanks for your quick response! I really appreciate your work :) It looks like staging is dropping packets again, for some devices I see gaps in the packet nonce and some sensors stopped completely :(

Is this still going on for your device? If yes, can you provide the Console device ID as well as how often does your device transfers data?

fishpepper, 2021-12-07:
Now its working. I just tested OTAA with a new sensor on staging. Thanks!

macpie, 2021-12-08:
Fix by https://github.com/helium/router/pull/546

closed 2021-12-08 by macpie