helium/router #533: Seeing 2 Join Request message in device debug view for first Join Request

leroyle, 2021-12-03:
1. Are you using Helium hosted Console or running your own open source Console?
hosted
2. If you are on Helium hosted Console, which url are you currently on? 
Both staging and production 
https://staging-console.helium.wtf  yesterday, but cannot connect today
 https://console.helium.com
3. If you are experiencing bugs in Helium hosted Console, please specify your Organization name and Device ID (ex. 3483050b-284d-47e2-8eda-e087fc03a753).
Device ID: a0b70b92-1f9a-45ac-98d5-e38711da95ea
Please describe your issue:

During the join process I'm seeing two join requests with in the console device debug view for a single join request from the device? 

For instance the two Join Request JSON messages have the following almost immediate time stamps:
 1st:   "reported_at": "1638484725038"
 2nd:  "reported_at": "1638484725227"
These are followed by the usual/expected Join Accept.
 
I've added some debug code and conditional break points within the Wisblock runtime time to ensure the runtime is not sending two messages, from what I can tell it is not, but I suppose the radio could be in some fashion.

I have a total of 3 Wisblock devices, they all exhibit the same unusual behavior.

I was seeing this yesterday (or day before) on staging console and the one time I tried on production. Today I verified this multiple times on production, I cannot connect to staging at the moment.

From the console: 

```
Join accept
Event Informationa few seconds ago

{
  "id": "5ccb423d-ad14-4601-83b6-64bc760c8716",
  "router_uuid": "a50e714d-d0e1-4794-89b8-fa4d46de8068",
  "category": "join_accept",
  "sub_category": "undefined",
  "description": "Join accept from AppEUI: 95612A1C2722EE26 DevEUI: 248024B8D303B4FD",
  "payload": "IMIq2mmko3CNjKPHway+jv8=",
  "payload_size": 17,
  "port": 0,
  "reported_at": "1638558620281"
}

Device Information

{
  "name": "WisBlock",
  "devaddr": "1D020048"
}

Hotspot

{
  "channel": 123,
  "frequency": 926.9,
  "id": "112DnKkS5qVvRoK1TFcbhT7ReezwtNTVmE2zTB2ZnAa3UMZYJ8kN",
  "lat": 32.9550615760182,
  "long": -96.75233778326856,
  "name": "harsh-sepia-gerbil",
  "rssi": 27,
  "snr": 0,
  "spreading": "SF10BW500"
}

Join request
Event Informationa few seconds ago

{
  "id": "8472e32f-dbcf-42d3-837d-4af9898b6ed4",
  "router_uuid": "cf487bb1-f027-4935-bce7-64d985f36a2a",
  "category": "join_request",
  "sub_category": "undefined",
  "description": "Join request from AppEUI: 95612A1C2722EE26 DevEUI: 248024B8D303B4FD",
  "payload": "",
  "payload_size": 0,
  "port": 0,
  "reported_at": "1638558618675"
}

Device Information

{
  "name": "WisBlock",
  "devaddr": "1E020048"
}

Hotspot

{
  "channel": 14,
  "frequency": 905.0999755859375,
  "id": "11ywkyiCU2BPPEAd6KZd8c2XpLdWiAYgEVa6B23uDgrgHcSHeyN",
  "lat": 32.95033353474442,
  "long": -96.74750285063357,
  "name": "festive-parchment-boar",
  "rssi": -121,
  "snr": -12.800000190734863,
  "spreading": "SF10BW125"
}

Join request
Event Informationa few seconds ago

{
  "id": "37fe3f21-aeb3-4fc7-87bd-6725c95b1423",
  "router_uuid": "cf487bb1-f027-4935-bce7-64d985f36a2a",
  "category": "join_request",
  "sub_category": "undefined",
  "description": "Join request from AppEUI: 95612A1C2722EE26 DevEUI: 248024B8D303B4FD",
  "payload": "",
  "payload_size": 0,
  "port": 0,
  "reported_at": "1638558618280"
}

Device Information

{
  "name": "WisBlock",
  "devaddr": "1D020048"
}

Hotspot

{
  "channel": 14,
  "frequency": 905.0999755859375,
  "id": "112DnKkS5qVvRoK1TFcbhT7ReezwtNTVmE2zTB2ZnAa3UMZYJ8kN",
  "lat": 32.9550615760182,
  "long": -96.75233778326856,
  "name": "harsh-sepia-gerbil",
  "rssi": -22,
  "snr": 13.199999809265137,
  "spreading": "SF10BW125"
}

```

jdgemm, 2021-12-10:
Thanks for feedback, pls test on the latest release and see if this is still an issue.

leroyle, 2021-12-11:
Still seeing it with Console V2.1.7 (production) , not during the first 5-6 joins, but did after that about 3-4 times in a row. 
Staging was not answering, will try later

leroyle, 2021-12-11:
I'm still seeing this with V2.1.8 on staging. 
As I sort of mentioned I'm not totally convinced it not an issue on the device end, sending multiple requests, but I think I've checked the lowest level radio code I have access to (I'll check that again).

Is there any way I can tell from my Helium original hotspot if it is seeing double join requests? I do not have access to an in the air analysis tool.

leroyle, 2021-12-12:
I will need to apologize. It looks like what I'm seeing is a second hot spot sending that second Join Request. I've been used to being the only hot spot on the block. Seems  two others have joined within shouting distance of my device.
It is plainly noted in the JSON message I attached to this issue as well as in the console had I expanded the events. the first request from festive-parchment-boar and the second from harsh-sepia-gerbil, mine.

I could have sworn when I first noticed this that I checked and they were both coming from my hot spot, but then again witnesses do tend to be unreliable.

Closing this issue. I'll reopen if I can confirm and capture a double request from the "same" hotspot.

closed 2021-12-12 by leroyle