helium/router #535: CFList Type 1 has wrong endianness

mkuyper, 2021-12-04:
It looks like the 16-bit channel mask values in the CFList my devices are receiving in US915 are not in little-endian format as required by the LoRaWAN specification:
```
PDU (enc): 20 3B42 5A36CF59 1606D739 A4 8E 75F7A66EFB3F2144F46FE5BBE31375B8 AF176C59
PDU (dec): 20 D35E C4486532 4C000048 08 00 FF000000000000000000000000000001 546D3D38

Join accept:
{'AppNonce': 12869331,
 'CFList': {'chmask': 'FF0000000000000000000000000000', 'type': 1},
 'DLSettings': 8,
 'DevAddr': 1207959628,
 'MHdr': 32,
 'MIC': 943549780,
 'NetID': 3302728,
 'RxDelay': 0,
 'statusMIC': True}
```
ChMask0 is thus `FF00` i.e. 0x00FF, enabling channels 0-7 and leaving the Helium channels of 8-15 disabled.

pvolodin, 2021-12-09:
And there seems to be no channel 65 ( 904.6 MHz / SF8 / BW500 )

mkuyper, 2021-12-09:
> And there seems to be no channel 65 ( 904.6 MHz / SF8 / BW500 )

Yes, that is correct. Not sure if that's intentional, but at least it's the same as is done via the LinkAdrReq command:
```
FOpts  : x0310000070031000FF00
         LinkADRReq<x03 u1(TXPow:0-3=0,DR:4-7=1) u2(ChMask:0-15=x0) u1(NbTrans:0-3=0,ChMaskCntl:4-6=7,RFU:7=0)>
         LinkADRReq<x03 u1(TXPow:0-3=0,DR:4-7=1) u2(ChMask:0-15=xFF00) u1(NbTrans:0-3=0,ChMaskCntl:4-6=0,RFU:7=0)>
```
The first command disables all channels (0-71), and the second enables channels 8-15.

pvolodin, 2021-12-09:
@mkuyper 
LinkADRReq controls datarate, CFList doesn't.
The sample you provided sets DR1, so there's no need in channel 65.

pvolodin, 2021-12-09:
@mkuyper  Can you please point me to the paragraph in LoRaWAN spec or regional settings doc which prohibits disabling the channels you don't need right now and in the foreseeable future? In US in most cases you cannot disable all 125 KHz channels as they shall be used for data rate back-off, but DR4 never can be DRNext.

mkuyper, 2021-12-09:
@pvolodin I fear that we are getting way off topic here in this issue, but the intention of the channel mask is to convey to the device all possible channels that can be used for uplink. There is nothing in the specification that forbids the LNS to dynamically enable and disable channels, but it limits the device's choices. In US915, if no 500 kHz channels are enabled, the device won't be able to transmit on DR4. Recall that a device is allowed at any time to make its own decision regarding datarate, and whether it wants to participate in ADR. For this to work meaningfully, the device needs to know the networks full capabilities. In general, the channels masks provided in the CFList and ones sent down in LinkAdrReq commands should match.

pvolodin, 2021-12-10:
> ...Not sure if that's intentional,...
>...
> The first command disables all channels (0-71), and the second enables channels 8-15.

A comment I foind in the sources:

>%% Region-specific Data Rate encoding table.
>%%
>%% TODO: bandwidth is not currently used because we're artificially
>%%       filtering the regional parameters for only 125kHz channels. Does
>%%       this make sense?

So probably that's really intentional.

mikev, 2021-12-17:
Thanks for the report.

The code and logs you posted appear correct.  The CFList array is displayed byte by byte.  Therefore the 2-byte chmask for channels 8-15 should be the most signficant bits and display as FF00.

Later you display a log of the mask: ChMask:0-15=xFF00, which is also a correct value.

I manually tested the code which computes the ChMask

>     Chans = [{8, 15}],
>     ChMaskTable = [
>         {2, mask, build_bin(Chans, {0, 15})},

This is computing the value 65,280 or 0xFF00 which is also correctly setting channels 8 to 15.

Are you not observing your device sending data on channels 8 to 15 between 903.9 and 905.3 MHz?

We'll investigate further if you believe this is not the case.

mkuyper, 2021-12-17:
> The code and logs you posted appear correct. The CFList array is displayed byte by byte. Therefore the 2-byte chmask for channels 8-15 should be the most signficant bits and display as FF00.

No, the CFList is not correct. It is true that the channel mask value for ChMask0 should be 0xFF00, but the correct little-endian representation of that value is `00FF`.

> Later you display a log of the mask: ChMask:0-15=xFF00, which is also a correct value.
That is the value that is sent down later in LinkAdrReq commands, which is correct.

> Are you not observing your device sending data on channels 8 to 15 between 903.9 and 905.3 MHz?
No, my device is sending on channels 0-7. I will personally vouch for the correctness of the device's behavior :wink:

pvolodin, 2021-12-17:
@mikev 
> The code and logs you posted appear correct.

let's take another look at the logs @mkuyper provided:

> 'CFList': {'chmask': '**FF00**00000000000000000000000000', 'type': 1},

and 

> FOpts  : x03100000700310**00FF**00

One of the masks marked with **bold** is incorrect, right?

mikev, 2021-12-18:
Fix will be deployed with release 66.5

mkuyper, 2021-12-20:
@mikev Thank you for tackling this issue. The CFList is now correct for devices on the staging console, but now the channel mask sent down in LinkADRReq commands has the wrong endianness (it was correct before):
```
PDU    : 600A0800488A010003000000700300FF000032B8A5CB

MHdr   : x60 DADN RFU:0 MAJOR:0
DevAddr: x4800080A/1207961610
FCtrl  : x8A ADREN:1 ADRARQ:0 ACK:0 MORE:0 OPTLEN=10
FCnt   : x0001/1
FOpts  : x03000000700300FF0000
         LinkADRReq<x03 u1(TXPow:0-3=0,DR:4-7=0) u2(ChMask:0-15=x0) u1(NbTrans:0-3=0,ChMaskCntl:4-6=7,RFU:7=0)>
         LinkADRReq<x03 u1(TXPow:0-3=0,DR:4-7=0) u2(ChMask:0-15=xFF) u1(NbTrans:0-3=0,ChMaskCntl:4-6=0,RFU:7=0)>
FPort  : ---/---
FRMPayl: --
MIC    : xCBA5B832/-878331854
```
This will have a huge impact on US915 devices if this hits production. This seems pretty critical to me, should I open a new issue for this?

mikev, 2021-12-20:
We are going to revert this change.  Our own hardware testing on stage shows this change causes incorrect behavior for both the Join and the LInkADDR reply.

Join Reply
CFList: Some([Frequency([0, ff, 0]), Frequency([0, 0, 0]), Frequency([0, 0, 0]), Frequency([0, 0, 0]), Frequency([0, 0, 0])])

LinkADDRReq Reply
LinkADRReq(LinkADRReqPayload([0, 0, 0, 70]))    LinkADRReq(LinkADRReqPayload([0, ff, 0, 0])) (edited)

Both should be 0xFF00

@mkuyper - Is it possible that your end-device is incorrectly parsing the CFList on the Join reply?  If you look at SemTech's reference implementaiton you'll note that a big-endian device must swap the bytes.  Here is the C code from SemTech's reference implementation:

        // Parse ChMask
        linkAdrParams->ChMask = ( uint16_t )payload[2];
        linkAdrParams->ChMask |= ( uint16_t )payload[3] << 8;

mkuyper, 2021-12-20:
> Our own hardware testing on stage shows this change causes incorrect behavior for both the Join and the LInkADDR reply.

Hmm, that's strange. But not necessarily surprising; there are a few borked stacks out there. What implementation are you testing with?

> Is it possible that your end-device is incorrectly parsing the CFList on the Join reply? If you look at SemTech's reference implementaiton you'll note that a big-endian device must swap the bytes. 

No, that's highly unlikely. I am using Basic MAC, the LMiC/TrackMAC successor which has successfully passed the LoRaWAN certification test suite in US915, and I have also manually parsed out the PDUs to verify. But here's the code from Semtech's LoRaMAC-Node that processes the CFList:

https://github.com/Lora-net/LoRaMac-node/blob/edff6ce161773e888687a9cd352ce2e2234e6b0d/src/mac/region/RegionUS915.c#L452-L453

```
        RegionNvmGroup2->ChannelsMask[chMaskItr] = (uint16_t) (0x00FF & applyCFList->Payload[cntPayload]);
        RegionNvmGroup2->ChannelsMask[chMaskItr] |= (uint16_t) (applyCFList->Payload[cntPayload+1] << 8);
```
As can be seen, the 16-bit mask is being read in little-endian order, i.e. the mask for channels 0-7 would be at offset `cntPayload`, and the mask for channels 8-15 would be at offset `cntPayload+1`.

mikev, 2021-12-21:
@mkuyper - We have put a possible fix for the join problem on stage.  Can you test your end-device against staging again?

If this fixes your problem, we'll go ahead and push a commit to fix this.

mkuyper, 2021-12-21:
@mikev Yes, it works now! :tada: 

Device Log:
```
[2021-12-20 19:38:26] lwm: TXSTART
[2021-12-20 19:38:26] TX[fcnt=-1,freq=904.7,sf=10,bw=125,pow=30,len=23]: 0000000000BBFFFFFF4101052BAAFFFFFF170D847A7415
[2021-12-20 19:38:26] lwm: TXDONE
[2021-12-20 19:38:31] RX[freq=925.7,sf=10,bw=500,rssi=-76,snr=7.00,len=33]: 20B5F20F87EF3DC5D9553C8629129DDEBC04ACC85F33FFA1CC6960CFD9D9E493E7
[2021-12-20 19:38:31] channel mask: ________ XXXXXXXX ________ ________ ________ ________ ________ ________ ________ 
[2021-12-20 19:38:31] lwm: JOINED
[2021-12-20 19:38:31] lwm: TXSTART
[2021-12-20 19:38:31] TX[fcnt=0,freq=905.3,sf=10,bw=125,pow=21,len=18]: 40080800488000000F59602153B6A0B80D23
[2021-12-20 19:38:32] lwm: TXDONE
[2021-12-20 19:38:33] RX[freq=927.5,sf=10,bw=500,rssi=-85,snr=7.50,len=22]: 60080800488A00000300000070030000FF0087A3172C
[2021-12-20 19:38:33] lwm: DATARATE
[2021-12-20 19:38:33] channel mask: ________ XXXXXXXX ________ ________ ________ ________ ________ ________ ________ 
[2021-12-20 19:38:33] lwm: TXCOMPLETE
```
Join Accept:
```
PDU: 20B5F20F87EF3DC5D9553C8629129DDEBC04ACC85F33FFA1CC6960CFD9D9E493E7

Join accept:
{'AppNonce': 12909039,
 'CFList': {'chmask': '00FF00000000000000000000000000', 'type': 1},
 'DLSettings': 8,
 'DevAddr': 1207961608,
 'MHdr': 32,
 'MIC': -1364451544,
 'NetID': 3302728,
 'RxDelay': 1,
 'statusMIC': True}
```
LinkAdrReq MAC command:
```
PDU: 60080800488A00000300000070030000FF0087A3172C

MHdr   : x60 DADN RFU:0 MAJOR:0
DevAddr: x48000808/1207961608
FCtrl  : x8A ADREN:1 ADRARQ:0 ACK:0 MORE:0 OPTLEN=10
FCnt   : x0000/0
FOpts  : x0300000070030000FF00
         LinkADRReq<x03 u1(TXPow:0-3=0,DR:4-7=0) u2(ChMask:0-15=x0) u1(NbTrans:0-3=0,ChMaskCntl:4-6=7,RFU:7=0)>
         LinkADRReq<x03 u1(TXPow:0-3=0,DR:4-7=0) u2(ChMask:0-15=xFF00) u1(NbTrans:0-3=0,ChMaskCntl:4-6=0,RFU:7=0)>
FPort  : ---/---
FRMPayl: --
MIC    : x2C17A387/739746695
```

Thank you again!

mikev, 2021-12-21:
@mkuyper - Excellent - that is good news.

The fix is here - https://github.com/helium/router/pull/555/files
We should be able to deploy this fix in this release.

closed 2021-12-21 by macpie

mikev, 2021-12-21:
Close Summary

Helium was returning an incorrectly encoded cflist (which includes chmask values) on the Join reply.  This issue was masked because we had implemented a work-around which also returned an empty list, i.e. multiple replies were returned for Join.

The work-around still exists and we'll remove it in another release, after we're sure this fix doesn't break any end-devices.

lthiery, 2021-12-21:
workaround #556