helium/router #549: AS923-3 frequencies marked as `incorrect_frequency` after PoC_v11 upgrade.

carlrakwireless, 2021-12-14:
After the POCv11, hotspots in the Philippines (AS923-3) suddenly got ALL invalid beacons and witnesses. There was no change in the frequency when the upgrade to POCv11 happened. It was still the the same AS923-3 uplink frequencies but those are now `incorrect_frequency`. Did we missed something in the config file or something related to that?

Here's the screenshot of the frequency and the invalid witnessing caused by `incorrect_frequency`.

[image]

I hope the helium team can have a look and make the fix soon.

shockool, 2021-12-14:
any update on this?

jdgemm, 2021-12-14:
Thanks for raising the issue, but it's the incorrect repository. Please raise your issue here: https://github.com/gradoj/hplans/issues

closed 2021-12-14 by jdgemm